BillyBlaze/OctoPrint-TouchUI #348: Layers not reporting in TouchUI to screen

samd12, 2020-01-02:
When using a plugin that reports back layer percentage in the dashboard it is not reporting the correct data to the touch interface.  What you are seeing in the browser is correct but not the same as on the touch screen.  Only the layer percent data.  Every thing else works perfectly.  See attached photo.  Yellow is computer and info is correct, red in the touch screen which does not move.  These photos were seconds apart
[touchui01](https://user-images.githubusercontent.com/4117470/71648386-918dd580-2cd1-11ea-91e1-04d63426bbd2.jpg)
[image]

samd12, 2020-01-17:
Guessing the project is on hold for a while?

BillyBlaze, 2020-02-01:
Sorry for the long delay, what plugin are you using?

BillyBlaze, 2020-03-08:
After investigating this issue; I found out that the 'Dashboard' plugins requires the GCodeViewer to be active to get the data. However for performance issues the GCodeViewer is disabled in 'touchscreen mode'. If you really want to see this data, then go to your TouchUI settings and disable Touchscreen mode.

commit cb21b298 (2020-03-08): :art: Disable layer progress in dashboard pluin

The dashboard plugin requires the GCodeViewer for displaying
layer progress; however in touchscreen mode this is disabled
and therefor the progress is always 0%

Now we hide the progess bar when using Touchscreen Mode

Fixes #348
closed 2020-03-11 by BillyBlaze