BillyBlaze/OctoPrint-TouchUI #350: No data updating on touch screen, Displaylayer error, fresh install

samd12, 2020-01-05:
I am completely updated on the Pi and in octopi.   Fresh install done today of everything.  I see my gauges and fields but there is no data.  I am also getting a message that "Cant get stats from DisplayLayerProgress"  Is it installed.  The plugin is installed.  I have even tried uninstalling/reinstalling with now luck.  I could not find a debug mode in the plugin please let me know what info you need
[image]
[image]

samd12, 2020-01-19:
Guessing this project is dead??

BillyBlaze, 2020-02-01:
Sorry for the long delay and thanks for reporting, Whats the name of the plugin?

samd12, 2020-02-02:
I have resolved everything except the layer does no update on the pi.  It just stays pinned at 100% even though on the computer it updates properly.  Octoprint TouchUI
[image]

BillyBlaze, 2020-03-08:
Great you fixed the data issue. Therefor I am closing this in favor of the other issue #348

closed 2020-03-08 by BillyBlaze