BillyBlaze/OctoPrint-TouchUI #353: Webcam icon not showing up in portrait mode

Mesmer88, 2020-01-12:
Webcam quick access button is not showing up in portrait mode. It appears in landscape.
I'm using a Samsung Galaxy Note 10 Plus phone to check Octoprint.

I see other users are having the same issue (not sure what they use to browse)

https://www.reddit.com/r/octoprint/comments/eha86s/how_to_get_webcam_quick_select_on_touch_ui_plugin/

https://imgur.com/a/EB2tMH7

[image]

Mesmer88, 2020-01-27:
UP for the bot.

BillyBlaze, 2020-02-01:
I think the icon with the padding just barely - doesn't - fit. The missing icon should be visible in the menu, or isn't this the case?

pebsi, 2020-02-24:
I get something similar. I have a different layout so for me it is the gcode viwer, consol and stop objects, which don't actually appear in the list or at the top of the screen in portrait mode and just controls don't appear in either.
I am using the addin themify, does this interfere with touch UI maybe? the themify snippet is below as well.
layout in website.
website view:
[image]

landscape mode
[image]
portrait mode
[image]

themify
[image]

commit aa3065a6 (2020-03-08): :art: Fine-tweak menu item width on mobile

Fine tweak the media queries for menu items
in the navbar. This will always show the
webcam item untill 300 pixels.

Fixes #353
closed 2020-03-11 by BillyBlaze