BillyBlaze/OctoPrint-TouchUI #364: UI does not boot - loads to consol

pebsi, 2020-02-11:
Hi, I have been trying to get this to work for a while now and so decided to log it here.
I have a pi 3b+ and I had the same error with both the stable and 1.4 RC of octoprint.

I followed all the install instructions on the wiki and the troubleshooting pages and everything appears to install fine but when I boot my pi with the screen attached it just boots to the consol with pi@octopi:~$ waiting for further inputs. 

The touchUI plugin is working because when I load the servers page on my phone it loads to touchUI. it is just the UI on the pi itself without touchscreen that is not working.

Am I missing something? what else can I provide you help (still relatively new with the consol.

Thanks

pebsi, 2020-02-15:
ok, on the stable version of octoprint I managed to get this working with a new install but only after also installing the desktop as per the code in intial SSH consol screen. after installing this and then changing the raspi-config to auto login CLI it loaded just fine.

However, now I have again moved to the 1.4RC4 and I get a new error which is 
Refused to connect.
same as issue #362

buzkill420, 2020-02-29:
> Hi, I have been trying to get this to work for a while now and so decided to log it here.
> I have a pi 3b+ and I had the same error with both the stable and 1.4 RC of octoprint.
>
> I followed all the [...]

Getting same issue, but without "waiting for inputs" and his solutions did not help. ie. cli settings in raspi-config

any help would be much appricated

BillyBlaze, 2020-03-07:
For anyone who doesn't have this fixed:

1. Stop TouchUI `sudo service touchui stop`
1. Run TouchUI manually `sudo xinit ~/TouchUI-autostart/chromium.xinit`
1. Paste the output here

BotiB, 2020-03-11:
OctoPrint loaded without TouchUI message in red and not starting up.

X.Org X Server 1.19.2
Release Date: 2017-03-02
X Protocol Version 11, Revision 0
Build Operating System: Linux 4.9.41-v7+ armv7l Raspbian
Current Operating System: Linux octopi 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20                                                                                                                                                              GMT 2018 armv7l
Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=600 bcm2708_fb.fbheight=                                                                                                                                                             1024 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000                                                                                                                                                               dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=aaab0fbb-02                                                                                                                                                              rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
Build Date: 18 October 2017  04:55:30PM
xorg-server 2:1.19.2-1+rpt1+deb9u2 (https://www.debian.org/support)
Current version of pixman: 0.34.0
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Wed Mar 11 19:05:51 2020
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
TouchUI: touch events enabled
TouchUI: screensaver enabled
wm_set_cursor_visibility: visible using XCreateFontCursor
Fontconfig warning: "/etc/fonts/fonts.conf", line 160: blank doesn't take any ef                                                                                                                                                             fect anymore. please remove it from your fonts.conf
[1576:1576:0311/190553.590637:ERROR:gl_surface_glx.cc(423)] glxQueryVersion fail                                                                                                                                                             ed
[1576:1576:0311/190553.590764:ERROR:gl_initializer_x11.cc(157)] GLSurfaceGLX::In                                                                                                                                                             itializeOneOff failed.
[1576:1576:0311/190553.598917:ERROR:viz_main_impl.cc(197)] Exiting GPU process d                                                                                                                                                             ue to errors during initialization
[1482:1517:0311/190553.614771:ERROR:browser_gpu_channel_host_factory.cc(120)] Fa                                                                                                                                                             iled to launch GPU process.
[1482:1707:0311/190553.839950:ERROR:bus.cc(394)] Failed to connect to the bus: C                                                                                                                                                             ould not parse server address: Unknown address type (examples of valid types are                                                                                                                                                              "tcp" and on UNIX "unix")
[1482:1517:0311/190554.210048:ERROR:browser_gpu_channel_host_factory.cc(120)] Fa                                                                                                                                                             iled to launch GPU process.
[1482:1482:0311/190554.228520:ERROR:gpu_process_transport_factory.cc(1029)] Lost                                                                                                                                                              UI shared context.
X.Org X Server 1.19.2
Release Date: 2017-03-02
X Protocol Version 11, Revision 0
Build Operating System: Linux 4.9.41-v7+ armv7l Raspbian
Current Operating System: Linux octopi 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20                                                                                                                                                              GMT 2018 armv7l
Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=600 bcm2708_fb.fbheight=                                                                                                                                                             1024 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000                                                                                                                                                               dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=aaab0fbb-02                                                                                                                                                              rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
Build Date: 18 October 2017  04:55:30PM
[... cut ...]                                                                                                                                  "tcp" and on UNIX "unix")
[1482:1517:0311/190554.210048:ERROR:browser_gpu_channel_host_factory.cc(120)] Fa                                                                                                                                                             iled to launch GPU process.
[1482:1482:0311/190554.228520:ERROR:gpu_process_transport_factory.cc(1029)] Lost

BillyBlaze, 2020-03-11:
Seems like its failing with GPU.

1. Run
    ```nano ~/TouchUI-autostart/chromium.xinit```
1. Add on line 90 at then end of the line the following:
    ```--disable-gpu```
1. Save and close (CTR-O, enter CTR-X)
1. Run `sudo xinit ~/TouchUI-autostart/chromium.xinit` if TouchUI loads congrats! else post output here again

BotiB, 2020-03-11:
Sorry but I'm not that much familiar with coding. How can I add that to line 90. I did this manual start after I logged in through SSH. It does not show line numbers.

BotiB, 2020-03-11:
chromium.xinit looks like this for me

#!/bin/sh

# DEFAULT VARIABLES
TOUCHUI_ARGS=""
TOUCHUI_DEBUG=""
TOUCHUI_EXTRA='echo "TouchUI: screensaver enabled"'

# Uncomment the next variable to generate a full log under ~/.config/chromium/chrome_debug.log
# or to enable the remote deugger, forward port by haproxy or by tunnel
#TOUCHUI_DEBUG="--remote-debugging-port=9222 --enable-logging --v=100"

# Load octoprint settings for getting $PORT
[ -r /etc/default/octoprint ] && . /etc/default/octoprint

# Load touchui settings for getting $TOUCHUI_USER and $FORCE_TOUCH
[ -r /etc/default/touchui ] && . /etc/default/touchui

# Remove restore last session after crash in Chrome
sed -i 's/"exited_cleanly":false/"exited_cleanly":true/' /home/$TOUCHUI_USER/.config/chromium/Default/Preferences
sed -i 's/"exit_type":"Crashed"/"exit_type":"None"/' /home/$TOUCHUI_USER/.config/chromium/Default/Preferences
# And another one for not minified Preferences (Chrome legacy)
sed -i 's/"exited_cleanly": false/"exited_cleanly": true/' /home/$TOUCHUI_USER/.config/chromium/Default/Preferences
sed -i 's/"exit_type": "Crashed"/"exit_type": "None"/' /home/$TOUCHUI_USER/.config/chromium/Default/Preferences

# Check if $FORCE_TOUCH is set to true
# This variable is set inside /etc/default/touchui
if [ "$FORCE_TOUCH" = true ]
then
        echo "TouchUI: touch events enabled (forced)"
        TOUCHUI_ARGS="$TOUCHUI_ARGS --touch-events=enabled"
else
        # Force Chrome to render with touch events
        # Bugfix #46
        TOUCHUI_DEVICE=$(xinput --list --long | grep XITouchClass)
        if [ -z "$TOUCHUI_DEVICE" ]
        then
                echo "TouchUI: touch events automatically detected"
        else
                echo "TouchUI: touch events enabled"
                TOUCHUI_ARGS="$TOUCHUI_ARGS --touch-events=enabled"
        fi
fi

# Check if we need to disable the screensaver
if [ "$DISABLE_SCREENSAVER" = true ]
then
        echo "TouchUI: screensaver disabled"
        TOUCHUI_EXTRA="xset s off & xset -dpms & xset s noblank"
fi

CHROME_VERSION=$(dpkg -s chromium-browser | grep "Version: " | sed -e 's/Version: //g')
CHROME_ARGS="   --no-first-run --kiosk $TOUCHUI_ARGS $TOUCHUI_DEBUG
                --dns-prefetch-disable --disable-sync-preferences --disk-cache-size=1048576
                --disable-java --disable-plugins --disable-extensions --disable-infobars
                --user-agent='TouchUI (X11, Chrome $CHROME_VERSION) (P:$PORT)'
                --start-maximized --window-position=0,0
                $(dirname "$0")/load-screen/startup.html"

				
[ -r /home/$TOUCHUI_USER/TouchUI-autostart/calibration.sh ] && . /home/$TOUCHUI_USER/TouchUI-autostart/calibration.sh

# Start window manager, unclutter to remove cursor and start chromium as specific user
eval $TOUCHUI_EXTRA &
matchbox-window-manager &
unclutter -idle 0.1 &
su $TOUCHUI_USER -c "chromium-browser $(echo $CHROME_ARGS|tr -d '\n')"

JMHaro, 2020-12-04:
I've been going though all the issue reporting trying to resolve my issues but this one didn't help either.  After I did do the last command above, I got the following messages:

pi@octopi2:~ $ sudo xinit ~TouchUI-autostart/chromium.xinit

X.Org X Server 1.20.4
X Protocol Version 11, Revision 0
Build Operating System: Linux 4.15.0-99-generic armv8l Raspbian
Current Operating System: Linux octopi2 5.4.72-v7+ #1356 SMP Thu Oct 22 13:56:54 BST 2020 armv7l
Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1 bcm2708_fb.fbwidth=480 bcm2708_fb.fbheight=320 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=6c586e13-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
Build Date: 27 August 2020  08:51:48AM
xorg-server 2:1.20.4-1+rpt2+deb10u1 (https://www.debian.org/support)
Current version of pixman: 0.36.0
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Fri Dec  4 01:37:46 2020
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
xinit: Unable to run program "xterm": No such file or directory
Specify a program on the command line or make sure that /usr/bin
is in your path.

xinit: connection to X server lost

waiting for X server to shut down (II) Server terminated successfully (0). Closing log file.

closed 2020-12-11 by github-actions[bot]