BillyBlaze/OctoPrint-TouchUI #365: touchui and octoprint 1.4 not working

TheMrCaveman, 2020-02-12:
I am currently testing octoprint in our farm and would like to use 1.4 betas on some devices but ran into the problem that touchui is not working anymore on 1.4 only works on 1.3 versions.

jansenchua, 2020-03-04:
I just upgraded to 1.4 and it's not working too... Just a white screen with error "refused to connect"

smokerjoe, 2020-03-04:
Same here. upgraded to 1.4 and it's not working too... Just a white screen with error "refused to connect"

revjkramer, 2020-03-04:
Same here just updated to 1.4 final today.

okdesuka63, 2020-03-04:
Same here and no backup :-(
Just revert to an older version, e.g. 1.3.12

`~/oprint/bin/pip install https://github.com/foosel/OctoPrint/archive/1.3.12.zip`
`sudo service octoprint restart`

[https://community.octoprint.org/t/how-can-i-revert-to-an-older-version-of-the-octoprint-installation-on-my-octopi-image/205](url)

TheNore, 2020-03-04:
here is the screen everyone is receiving (me included) on final 1.40 octoprint
[image]

roelleke, 2020-03-04:
After today upgrade to 1.4 I have the same problem.

floridaservices, 2020-03-05:
I'm not sure I would have ever seen this issue beforehand as it's only touchui that's affected, but I had to do the version reversion trick detailed by okdesuka63 (thanks!) to get my touchscreen back as well.  Easiest way is to SSH in, can't do it from the GUI and it's a pain to do it from the touch screen terminal.

diablex2, 2020-03-05:
Same problem here, 
After upgraded to 1.4.0 Touch UI is not loading.
Any solutions?

rogodra, 2020-03-05:
Same problem after upgrading to 1.4.0.
[image]

BillyBlaze, 2020-03-05:
Thanks for reporting. I'll have a look at this as soon as possible. Closing this duplicated issue, please follow #368

closed 2020-03-05 by BillyBlaze

cappy0815, 2020-03-06:
Hi,

got the same problem since upgrading to 1.4

> nano TouchUI-autostart/chromium.xinit
> 
> change and comment:
> 
> http://localhost:80/"
> #http://localhost:$TOUCHUI_PORT/"
> 
> sudo service touchui restart

I have no lines like this or any symilar entries ...
Wondering why!?

Cappy

BillyBlaze, 2020-03-06:
Don't do this. To Fix your issue follow this steps instead: https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/368#issuecomment-595679383

cappy0815, 2020-03-06:
Hi,
I checked your recommended feed.
First steps did not work ...

Finally I was working on the same lines as here in this threat.
I changed values to ...:80 and it is working.

Cappy