BillyBlaze/OctoPrint-TouchUI #370: OctoPrint Crashes every time and immediately after running "sudo ~/TouchUI-autostart/helpers/install"

designcafe3d, 2020-03-06:
Hello,

I have successfully installed 5 or 6 Octoprint instances. This one is on a Raspberry pi 4B. And, I've already had this fully working. However, I had originally installed the "light desktop" on the first installation and wanted to get rid of it. Long story short I wound-up having to just start from scratch to get this accomplished. Now, I'm on my FIFTH re-install of Octoprint on this same machine that worked beautifully before because, now, I can get all the way through all of the installation including getting the 4" touchscreen installed... then... when I go to install "Touch UI" by running the "sudo ~/TouchUI-autostart/helpers/install" command... after if finishes all the processes and reboots, Octoprint now crashes and gives me this message. 

---------------------------------
The OctoPrint server is currently not running
If you just started up your Raspberry Pi, please wait a couple of seconds, then try to refresh this page.

If the issue persists, please log into your Raspberry Pi via SSH and check the following:

Verify that the process is running: ps -ef | grep -i octoprint | grep -i python should show a python process:
pi@octopi:~ $ ps -ef | grep -i octoprint | grep -i python
pi 1441 1 6 11:12 ? 00:00:15 /home/pi/oprint/bin/python
/home/pi/oprint/bin/octoprint --host=127.0.0.1 --port=5000
If it isn't, the question is why. Take a look into ~/.octoprint/logs/octoprint.log, there might be an error logged in there that helps to determine what's wrong.
You might also want to try if you can restart the server (if no obvious error is visible): sudo service octoprint restart.
If all that doesn't help to trouble shoot the issue, you can seek support on the OctoPrint Community Forum. Please provide your OctoPi and OctoPrint versions as well as your octoprint.log and explain what you already tried and observed as detailed as possible.
------------------------------

I just don't know what's going wrong at this point considering all of this was working just yesterday. And, I reinforce, I'm on my fifth re-install of OctoPrint and the TouchUI and the failure happens at the same place everytime.... immediately after I run "sudo ~/TouchUI-autostart/helpers/install". 

HELP! Any advice for help you could provide would be greatly appreciated. 

Thank you so much for your time. 

Best,
Doug

Egalistel, 2020-03-06:
I am in the same case, I carried out the last update of octoprint and following that the touch no longer works, I carried out the same manipulation and I finally had to come back on a previous version of octoprint for the time being.

HologramDK, 2020-03-06:
I'm installing TouchUI for the first time and experience the same issue.

Starting octoprint in safemode yields this message:

> pi@octopi:~/.octoprint $ ~/oprint/bin/octoprint safemode
> Error parsing the configuration file, it appears to be invalid YAML. The parser reported an error on line 3, column 8.
> There was a fatal error initializing the settings manager.

Removing the accessControl section inserted by TouchUI fixes the issue with octoprint not starting, however, now TouchUI is just launching into a "sad guy" page not found like icon.

HologramDK, 2020-03-06:
Never mind, I can see that the reason has been found and a workaround is available here:
https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/368#issuecomment-595820612

BillyBlaze, 2020-03-07:
Thanks for reporting. A fix in the bootloader is coming up.

BillyBlaze, 2020-03-07:
This issue have been fixed in the bootloader, for anyone who currently have this issue, fix it by doing the following:

Run `nano ~/.octoprint/config.yaml` and check if the indentention is still correct:

The config should like this:
```
accessControl:
    autologinAs: test
    autologinLocal: true
    salt: foo
```

and not like this:
```
accessControl:
    autologinAs: test
    autologinLocal: true
      salt: foo
```

closed 2020-05-25 by BillyBlaze