BillyBlaze/OctoPrint-TouchUI #375: TouchUI won't boot up after update to 1.4.0

BotiB, 2020-03-11:
Hello,
Just updated Octoprint to 1.4.0 released recently. After update, TouchUI won't start anymore.
Gets to the screen where says connecting to OctoPrint, than after few seconds the logo changes to red and says OctoPrint loaded without TouchUI.
I did disable the security feature mentioned by Foosel, but still the same.
Any one else experiencing the same issue?
Thanks

BillyBlaze, 2020-03-11:
Thanks for reporting, is OctoPrint running in safe mode? and are you sure that TouchUI is not disabled?

BotiB, 2020-03-12:
TouchUI is not disabled and OctoPrint isn't running in safe mode either.

nsmdigitalpc, 2020-03-21:
I Reinstalled and now its working again
cd ~
sudo apt update
sudo apt install python-pip python-dev python-setuptools python-virtualenv git libyaml-dev build-essential
mkdir OctoPrint && cd OctoPrint
virtualenv venv
source venv/bin/activate
pip install pip --upgrade
pip install octoprint

OutspokenPlanet, 2020-03-24:
@nsmdigitalpc Thank you, followed your steps, worked like a charm

tbojim1975, 2020-03-25:
You are welcome :-)

closed 2020-04-15 by stale[bot]