BillyBlaze/OctoPrint-TouchUI #376: Typo in available version

speendo, 2020-03-11:
I just updated to the latest version **0**.3.14 of your magnificent plugin.

However, after updating the "new version popup" still appears and tells me to update to version **1**.3.14. 

I think, one of these two sources has to be changed:
* https://github.com/BillyBlaze/OctoPrint-TouchUI/releases/tag/1.3.14
* https://github.com/BillyBlaze/OctoPrint-TouchUI/blob/1.3.14/setup.py#L6

Thank you anyways for providing this plugin!

[image]

TheNore, 2020-03-11:
2nd this

xkravinx, 2020-03-12:
I am experiencing this also.

BillyBlaze, 2020-03-12:
Ouch, that is a painful mistake. I have updated the release and removed the previous one.

Are you able to update now?

speendo, 2020-03-12:
I had to force clean cache in search for updates (can you follow me :D) but now it's working alright!

Nice Commander Keen image, btw :+1:

xkravinx, 2020-03-12:
Would you mind sharing the steps to clean cache?  I still cannot update and I'm receiving an error message now when I try.  Thank you!

speendo, 2020-03-12:
Sure, no problem!

Just follow these steps 1-3
[image]

Sorry, my UI is in German, however, I am sure you can still figure it out!

pebsi, 2020-03-12:
It wont let me update and says below error

ERROR: HTTP error 404 while getting https://github.com/BillyBlaze/OctoPrint-TouchUI/archive/1.3.14.zip
ERROR: Could not install requirement https://github.com/BillyBlaze/OctoPrint-TouchUI/archive/1.3.14.zip because of error 404 Client Error: Not Found for url: https://codeload.github.com/BillyBlaze/OctoPrint-TouchUI/zip/1.3.14
ERROR: Could not install requirement https://github.com/BillyBlaze/OctoPrint-TouchUI/archive/1.3.14.zip because of HTTP error 404 Client Error: Not Found for url: https://codeload.github.com/BillyBlaze/OctoPrint-TouchUI/zip/1.3.14 for URL https://github.com/BillyBlaze/OctoPrint-TouchUI/archive/1.3.14.zip

BotiB, 2020-03-12:
For me the update worked, but having the same issue like before. When it boots up, logo turns to red and saying OctoPrint loaded without TouchUI.

X.Org X Server 1.19.2
Release Date: 2017-03-02
X Protocol Version 11, Revision 0
Build Operating System: Linux 4.9.41-v7+ armv7l Raspbian
Current Operating System: Linux octopi 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l
Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=600 bcm2708_fb.fbheight=1024 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=aaab0fbb-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
Build Date: 18 October 2017  04:55:30PM
xorg-server 2:1.19.2-1+rpt1+deb9u2 (https://www.debian.org/support)
Current version of pixman: 0.34.0
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Thu Mar 12 17:49:11 2020
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
TouchUI: touch events enabled
TouchUI: screensaver enabled
wm_set_cursor_visibility: visible using XCreateFontCursor
Fontconfig warning: "/etc/fonts/fonts.conf", line 160: blank doesn't take any effect anymore. please remove it from your fonts.conf
[1526:1526:0312/174913.705520:ERROR:gl_surface_glx.cc(423)] glxQueryVersion failed
[1526:1526:0312/174913.705699:ERROR:gl_initializer_x11.cc(157)] GLSurfaceGLX::InitializeOneOff failed.
[1526:1526:0312/174913.712371:ERROR:viz_main_impl.cc(197)] Exiting GPU process due to errors during initialization
[1428:1463:0312/174913.718820:ERROR:service_manager_context.cc(252)] Attempting to run unsupported native service: /usr/lib/chromium-browser/content_gpu.service
[1428:1463:0312/174913.728223:ERROR:browser_gpu_channel_host_factory.cc(120)] Failed to launch GPU process.
[1428:1653:0312/174913.882117:ERROR:bus.cc(394)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[1428:1463:0312/174914.234820:ERROR:browser_gpu_channel_host_factory.cc(120)] Failed to launch GPU process.
[1428:1428:0312/174914.242693:ERROR:gpu_process_transport_factory.cc(1029)] Lost UI shared context.

TheNore, 2020-03-12:
this was fixed for me by the following steps

-Completely remove touchUI Octoprint plugin, and settings
-Clear plugin cache
-Reboot whole SYSTEM not just octoprint
-Reinstall TouchUI plugin in octoprint
-reboot whole SYSTEM not just octoprint
-Should be working as intended if underlying backend is setup and working.

BotiB, 2020-03-12:
I just tried these steps, but unfortunately did not work for me. Having the same issue.
On PC browser can switch to TouchUI and its fine, but on printers touchscreen won't boot up.

closed 2020-04-02 by stale[bot]