BillyBlaze/OctoPrint-TouchUI #377: 0.3.14 options to "Proceed" or "cancel" are unable to be pressed

Kokorone, 2020-03-13:
Having a print running, then selecting cancel has a prompt come up that previously had buttons for proceed or cancel. Both of these buttons are hidden from view, somewhere below the nav bar. I've tested with chrome, edge, and safari on my iphone and all of them have the buttons missing.

[image]

ordubis, 2020-03-22:
I just want to confirm I'm having this issue as well.

Tried both the latest and maintenance branches of TouchUI, OctoPrint 1.4.0 running on OctoPi 0.17.0, phone is an iPhone X running iOS 13.4.

Does not show the Proceed or Cancel buttons on portrait mode, but does on landscape, and I hate using landscape so I'd love if this were fixed. Considering there was another ticket opened a while back with the same issue, it'd be nice if the fancy bottom-left floating buttons would just get moved to directly under the prompt. If nothing else, it makes it closer to where a thumb (when using a phone) would reach.

ordubis, 2020-03-22:
For the current version of TouchUI, I found out tonight that removing `min-height:100vh;` from `#touch body .modal-scrollable .modal` in _octoprint_touchui\static\css\touchui.css_ allows the confirmation box to show up at a reasonable size, which then displays the Proceed and Cancel buttons; it honestly looks better that way too. (Leaving this in both support tickets on this issue in hopes that the plugin author sees it.)

barrettdent, 2020-03-28:
I am also having this issue.  This issue makes TouchUI essentially useless.  I cannot restart my server.  I cannot shutdown my server.  I cannot cancel a print.

dchauran, 2020-04-02:
@ordubis I have seen several reports of this issue, and yours is the only one with a fix. Would you mind submitting a pull request? That way others who have the issue can hopefully find this and make the change on their servers more easily.

ordubis, 2020-04-02:
> 
> 
> @ordubis I have seen several reports of this issue, and yours is the only one with a fix. Would you mind submitting a pull request? That way others who have the issue can hopefully find this and make the change on their servers more easily.

I'll be honest, I don't know how to git really well. I downloaded as a zip, tinkered, and uploaded to my pi via the page. If you or anyone else would like to with the change, I'm cool with it, I just wanted to ensure others knew the solution existed :)

tedder, 2020-04-05:
Discovered this problem with Android today. Screenshot from a Pixel 3.

[image]

vicwomg, 2020-04-05:
just chiming in that until this is properly fixed, you can also toggle full screen UI 
Menu > TouchUI settings > Toggle fullscreen
In this mode, there is enough screen real estate to show the confirmation buttons.

If you're on Chrome/Android, you can also scroll down slightly until the browser URL bar is hidden, then press the stop button, this achieves a similar effect.

dchauran, 2020-04-06:
Personally I went in and tweaked the css to fix it until the fix is in the release.

tedder, 2020-04-23:
how dare you bot, this is still desperately needing a fix.

BillyBlaze, 2020-07-21:
I looked into this issue, however I am unable to reproduce this issue on my android device.

That aside, I removed all declarations with `100vh` and refactored it a bit so that the common issue with the navigation bar is gone. Can somebody confirm this for me that it also worked for them? 

1. Uninstall TouchUI
1. Install TouchUI with the link:
    `https://github.com/BillyBlaze/OctoPrint-TouchUI/archive/maintenance.zip`

tedder, 2020-07-21:
Just did this on stable octoprint (1.4.0) using my stock Pixel 3 phone, the 'cancel' fits now. Thanks.

commit 94946c78 (2020-07-21): :bug: Attempt to fix modal buttons

On some devices you cannot scroll towards the buttons, therefor making it
impossible to save the modal.

Fixes #377