BillyBlaze/OctoPrint-TouchUI #381: TouchUI Screen Won't Load

Toxicl16, 2020-03-20:
Hi,
    I followed the steps on having the TouchUI boot to browser automatically. However, whenever I boot the interface never loads. The only thing that happens is the raspberry pi only loads to the console from booting. This is similar to issue #364 I added --disable-gpu to line 90. However, this too doesn't fix the problem. Please and thank you for the help,
ToxicL

TheNore, 2020-03-20:
ok you will have to go into raspi-config and change your boot to
preferences.

Toxicl16, 2020-03-20:
I have it set to autologin console. The option of desktop says I have to install lightdm. What do I need to change it to?

TheNore, 2020-03-20:
that is actually correct, did you try pulling the Git repository again? and
then the easy installer just to make sure. Its Easy installer 0.0.3 ( I
think)

Toxicl16, 2020-03-20:
After some time of digging, I was able to find a post on a forum. Now, the issue is fixed. The post is linked below. However, my screen is touch-screen but when pressing buttons on the interface the interface doesn't register them.

https://www.raspberrypi.org/forums/viewtopic.php?t=162323

TheNore, 2020-03-20:
are your touchscreen drivers installed correctly?

Toxicl16, 2020-03-20:
I believe my screen is faulty. It has a minor crack because i had dropped it a few weeks ago. Also, I tried it on another version of raspbian and it didn't work. Thanks for the prompt responses.

closed 2020-04-11 by stale[bot]