BillyBlaze/OctoPrint-TouchUI #384: After recent update, screen size not right

3DEnthusiast, 2020-03-21:
After the most recent update, when I cancel a print I can not see the confirm button at the bottom. TouchUI will not allow me to scroll up or down to see it.

dchauran, 2020-03-25:
Same issue here, I have to switch to full screen to see it.

3DEnthusiast, 2020-03-25:
I have actually figure out that if you scroll the page down, you can see the submit button. It's a little bit of a pain but works.

emilecantin, 2020-03-26:
I'd bump this up to CRITICAL: I had to cancel a print a few minutes ago, and I couldn't click on the confirm button! I had to run to my desktop computer on the other side of the house to cancel properly.

bushibot, 2020-03-28:
if you rotate a IOS device horizontal then you can see the yes cancel but yeah this messed up currently. The cancel screen floating box over fills the vertical screen.

timhere, 2020-04-05:
Same issue here, I can't find a way to confirm prompts.

CodeBrauer, 2020-04-06:
Maybe there is a way to replace the built-in prompts with native `prompt()`? This would be much easier and more accessible for any device?

bushibot, 2020-04-06:
It's dumb, but if I unlock rotation, and turn the device horizontally I can then access the prompts fine, the screen will slide up a bit and reveal them. Can't do this in vertical mode even with scrolling on. So, it's kind of a work around. Though way to many clicks for those of us who keep our phone rotation locked.

3DEnthusiast, 2020-04-06:
> I'd bump this up to CRITICAL: I had to cancel a print a few minutes ago, and I couldn't click on the confirm button! I had to run to my desktop computer on the other side of the house to cancel properly.

I'm not sure I know how to bump it.

BillyBlaze, 2020-07-21:
I looked into this issue, however I am unable to reproduce this issue.

That aside, I removed all declarations with 100vh and refactored it a bit so that the common issue with the navigation bar is gone. Can somebody confirm this for me that it also worked for them?

1. Uninstall TouchUI
1. Install TouchUI with the link:
    `https://github.com/BillyBlaze/OctoPrint-TouchUI/archive/maintenance.zip`

Prizmagnetic, 2020-07-28:
@BillyBlaze I just tested your fix and it has resolved the issue for me.  I'm using a Samsung Galaxy S9 with chrome

BillyBlaze, 2020-08-04:
Awesome thanks for reporting. This fix has been released in latest version (0.3.15)

closed 2020-08-04 by BillyBlaze

Prizmagnetic, 2020-08-26:
the issue has come back, now that I'm running 0.3.15 instead of the maintenance branch