BillyBlaze/OctoPrint-TouchUI #386: New version of numberkeyboard is broken

kazolar, 2020-03-27:
Has a weird -> next to backspace, makes it very difficult to press backspace
[image]

klsc18, 2020-04-03:
same here. I can not delete anything with the x-button, only changing temperature with up and down in textfield works. this issue is only after update to the new version.

kazolar, 2020-04-04:
I found the code and deleted the arrow keys in the JavaScript library, and that worked. I don't know how to fix it so I can reinstall the plug-in

PietsHost, 2020-04-26:
> 
> 
> I found the code and deleted the arrow keys in the JavaScript library, and that worked. I don't know how to fix it so I can reinstall the plug-in

Could you specify where to find the code and what piece of code did you delete?

CCSnell, 2020-06-03:
Would definitely like to know also, having the same issue. I'm running in portrait mode, not sure if it makes a difference, just thought I'd mention.

lciscon, 2020-06-22:
I have encountered the same issue.

lciscon, 2020-06-22:
I believe it is an issue with the definition of the number keyboard in keyboard.js.  I *think* it should not have the {left}  and {right} keys in the second row.  I'm trying to figure out how to remove those from an active install and try it:

```
		number: {
			display: {
				'bksp'   :  ' ',
				'a'      :  'Save',
				'c'      :  'Cancel'
			},

			layout: 'custom',
			customLayout: {
				'default' : [
					'{bksp} 1 2 3 4 5 6 7 ',
					'{accept} {c} {left} {right} 8 9 0 - , . '
				]
			},
		}
```

lciscon, 2020-06-23:
I've confirmed that by changing this row:

```
'{accept} {c} {left} {right} 8 9 0 - , . '
```

to this:

```
'{accept} {c} 8 9 0 - , . '
```

it fixes the issue.

BillyBlaze, 2020-07-21:
Thanks all for reporting. This issue has been fixed on the maintenance branch and will be released soon.

commit c9b6deac (2020-07-21): :bug: Fix virtual keyboard number input

Chrome no longer allows text selection on input types and kills the
virtual keyboard

Fixes #386 #400

:art: Add padding to page when opening keyboard

Allow some extra space on the page to allow to view the input field