BillyBlaze/OctoPrint-TouchUI #394: Touch UI Couln't to Connect on PI Touch Screen

dsrq, 2020-04-10:
After Octoprint Last Update on my Raspberry PI 4 with 3.5in Touchscreen when the PI is booting im getting the Touch Ui connecting screen on the touchscreen, after few seconds ive got a white screen with a little sad page emoticon .  REFUSED TO CONNECT .    Everything was working #1 before the Octoprint Update.  
Running octoprint 1.4.0 on octopi 0.17.0    and Touch UI 0.3.14.      

Touch UI works on my laptop  but not directly on my 3.5in touchscreen.

Thanks !

devhammer, 2020-04-12:
See this workaround:

https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/368#issuecomment-595036339

I actually modified the file **/etc/default/touchui**, using this command:

`sudo nano /etc/default/touchui`

I replaced "localhost" in the TOUCHUI_URL variable value with the IP address assigned to my Pi (I set that as a static IP in my router so it's always the same), and saved and rebooted, and that sorted the connection issue for me. I also disabled the screensaver later in the file [per these instructions](https://howchoo.com/g/ywrlzdk2ndj/disable-touchui-screensaver).

dsrq, 2020-04-12:
Thanks ! @devhammer   ive followed the end of the #368 ,   ive did a Git pull of the Bootloader, and everything works as normal ! Thanks !

closed 2020-05-04 by stale[bot]