BillyBlaze/OctoPrint-TouchUI #396: strange screen after boot up like permision denied

freezeghost, 2020-04-14:
Hello I just have strange screen like permision denied solved in #368 
This screen is showed after aprox 1 min after start up. If I restart service it will show again :-(
[image]

freezeghost, 2020-04-15:
I try to solve this issue. And it is not related with TouchUI but with chromium browser. I try install latest version of chromium on other rpi and after same time (always 57 sec, strange :-) ) is same issue.
Has some one same issue?

freezeghost, 2020-04-19:
trouble disappear when cleared chromium`s cache with "rm -rf ~/.cache/chromium"

TheNore, 2020-04-20:
thanks for this, happened to me too - would a simple update help as well?

closed 2020-05-13 by stale[bot]