BillyBlaze/OctoPrint-TouchUI #397: No on-screen keyboard

Jozer99, 2020-04-17:
Fresh install of Octopi 0.17 on a Pi 3B with Raspberry Pi official touchscreen.  When I install Boot to Browser via the install script, everything seems to go fine.  When I use the touchscreen, entering a text box does not bring up the soft-keyboard.  I can't modify or enter any text.

Jozer99, 2020-04-17:
Found a manual fix; on the Pi screen, go to the menu and TouchUI settings, and toggle the onscreen keyboard to active (inactive by default).  On older versions, this was configured automatically for the local browser session.

Jozer99, 2020-04-18:
Update: this fix needs to be re-applied on every reboot of the pi.  Is there a way to make this auto-configured by the Boot-to-Browser installer, or at least persistent across reboots?

lciscon, 2020-04-22:
I've encountered this issue as well. Fresh install.  This is a critical issue.

lciscon, 2020-04-22:
It looks like the cookie data is not being saved for some reason.

Jozer99, 2020-05-06:
This issue has not been addressed, please fix!

lciscon, 2020-05-24:
Could it be related to the fact that the CONFIG_DIR line appears to be commented out by default in the touchui.default file? 

# Path for the profile - defaults to ram disk to reduce writes
#CONFIG_DIR=/run/touchui-chromium/

lciscon, 2020-05-24:
Actually I think it is because the /run directory is set as a tmpfs, so it is stored in RAM rather than the SD.  changing the CONFIG_DIR to something that is not set as a tmpfs should fix it.  I'm going to test this this afternoon.

lciscon, 2020-05-25:
I have confirmed that that is the issue.  If you Change the CONFIG_DIR value in chromium.xinit to point to a directory that is not RAM-based it works fine.  /tmp and /run are both ram based by default (at least in my Raspberry PI configuration).  

For example if you change it to this:

CONFIG_DIR=/home/pi/tmp

Then the TouchUI settings persist between reboots.

BillyBlaze, 2020-07-21:
Thanks for reporting, this issue is fixed on the maintenance branch and will be released soon. You might need to restart your Pi after installing the next update.

commit f4f6f978 (2020-07-21): :bug: Force virtual keyboard with bootloader

Fixes #397