BillyBlaze/OctoPrint-TouchUI #403: TouchUI doesn't start untill i install the desktop for OctoPi

Pickass1988, 2020-05-11:
Hey, i have an Error i can't find anything on google for it. 
I've installed OctoPi on several PI's with the same touchscreen and TouchUI wont start untill i install the desktop version. 

It might be a problem with the driver of my screen, but i am completely unexperienced. 

I've tried several things including the
sudo xinit /home/pi/OctoPrint-TouchUI-autostart/start_chromium.xinit

If i do that it gives me the following errors: 

X.Org X Server 1.20.4
X Protocol Version 11, Revision 0
Build Operating System: Linux 4.15.0-48-generic armv8l Raspbian
Current Operating System: Linux octopi 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l
Kernel command line: coherent_pool=1M 8250.nr_uarts=1 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:12:29:D4 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=tty1 console=ttyS0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait fbcon=map:10 fbcon=font:ProFont6x11 logo.nologo
Build Date: 05 June 2019  12:49:54PM
xorg-server 2:1.20.4-1+rpt1 (https://www.debian.org/support)
Current version of pixman: 0.36.0
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Mon May 11 20:16:48 2020
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
(EE)
Fatal server error:
(EE) no screens found(EE)
(EE)
Please consult the The X.Org Foundation support
         at http://wiki.x.org
 for help.
(EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
(EE)
(EE) Server terminated with error (1). Closing log file.
xinit: giving up
xinit: unable to connect to X server: Connection refused
xinit: server error

What did i do wrong?

ChristopherRush, 2020-05-12:
I'm having the same issue.

BillyBlaze, 2020-05-13:
Thanks for reporting

1. what touchscreen are we talking about?
2. what is the output of this command `sudo ls /dev/`
3. what is the output of this command `sudo cat /usr/share/X11/xorg.conf.d/99-fbdev.conf`

Pickass1988, 2020-05-13:
Hey,

The screen is called sain smart 5 inch on Amazon. 
I'll post the outputs as soon as possible. I am at the hospital till Monday. 

Thank you in advance

Pickass1988, 2020-05-18:
Here are the outputs:
With Desktop installed: 

sudo ls /dev/

autofs           loop-control        ram9     tty26  tty54      vcs7
block            mapper              random   tty27  tty55      vcsa
btrfs-control    media0              raw      tty28  tty56      vcsa1
bus              mem                 rfkill   tty29  tty57      vcsa2
cachefiles       memory_bandwidth    serial0  tty3   tty58      vcsa3
char             mmcblk0             serial1  tty30  tty59      vcsa4
console          mmcblk0p1           shm      tty31  tty6       vcsa5
cpu_dma_latency  mmcblk0p2           snd      tty32  tty60      vcsa6
cuse             mqueue              stderr   tty33  tty61      vcsa7
disk             net                 stdin    tty34  tty62      vcsm
fb0              network_latency     stdout   tty35  tty63      vcsm-cma
fd               network_throughput  tty      tty36  tty7       vcsu
full             null                tty0     tty37  tty8       vcsu1
fuse             ppp                 tty1     tty38  tty9       vcsu2
gpiochip0        ptmx                tty10    tty39  ttyAMA0    vcsu3
gpiochip1        pts                 tty11    tty4   ttyprintk  vcsu4
gpiochip2        ram0                tty12    tty40  ttyS0      vcsu5
gpiomem          ram1                tty13    tty41  uhid       vcsu6
hwrng            ram10               tty14    tty42  uinput     vcsu7
initctl          ram11               tty15    tty43  urandom    vhci
input            ram12               tty16    tty44  v4l        video10
kmsg             ram13               tty17    tty45  vchiq      video11
log              ram14               tty18    tty46  vcio       video12
loop0            ram15               tty19    tty47  vc-mem     watchdog
loop1            ram2                tty2     tty48  vcs        watchdog0
loop2            ram3                tty20    tty49  vcs1       zero
loop3            ram4                tty21    tty5   vcs2
loop4            ram5                tty22    tty50  vcs3
loop5            ram6                tty23    tty51  vcs4
loop6            ram7                tty24    tty52  vcs5
loop7            ram8                tty25    tty53  vcs6

sudo cat /usr/share/X11/xorg.conf.d/99-fbdev.conf

cat: /usr/share/X11/xorg.conf.d/99-fbdev.conf: No such file or directory

Then i did a clean install and without desktop installed: 

sudo ls /dev/

autofs           loop7               ram7     tty23  tty50      vcs2
block            loop-control        ram8     tty24  tty51      vcs3
btrfs-control    mapper              ram9     tty25  tty52      vcs4
bus              media0              random   tty26  tty53      vcs5
cachefiles       mem                 raw      tty27  tty54      vcs6
char             memory_bandwidth    rfkill   tty28  tty55      vcsa
console          mmcblk0             serial0  tty29  tty56      vcsa1
cpu_dma_latency  mmcblk0p1           serial1  tty3   tty57      vcsa2
cuse             mmcblk0p2           shm      tty30  tty58      vcsa3
disk             mqueue              snd      tty31  tty59      vcsa4
fb0              net                 stderr   tty32  tty6       vcsa5
fd               network_latency     stdin    tty33  tty60      vcsa6
full             network_throughput  stdout   tty34  tty61      vcsm
fuse             null                tty      tty35  tty62      vcsm-cma
gpiochip0        ppp                 tty0     tty36  tty63      vcsu
gpiochip1        ptmx                tty1     tty37  tty7       vcsu1
gpiochip2        pts                 tty10    tty38  tty8       vcsu2
gpiomem          ram0                tty11    tty39  tty9       vcsu3
hwrng            ram1                tty12    tty4   ttyAMA0    vcsu4
initctl          ram10               tty13    tty40  ttyprintk  vcsu5
input            ram11               tty14    tty41  ttyS0      vcsu6
kmsg             ram12               tty15    tty42  uhid       vhci
log              ram13               tty16    tty43  uinput     video10
loop0            ram14               tty17    tty44  urandom    video11
loop1            ram15               tty18    tty45  v4l        video12
loop2            ram2                tty19    tty46  vchiq      watchdog
loop3            ram3                tty2     tty47  vcio       watchdog0
loop4            ram4                tty20    tty48  vc-mem     zero
loop5            ram5                tty21    tty49  vcs
loop6            ram6                tty22    tty5   vcs1

sudo cat /usr/share/X11/xorg.conf.d/99-fbdev.conf
cat: /usr/share/X11/xorg.conf.d/99-fbdev.conf: No such file or directory

The driver for the screen can be found at: 
https://github.com/goodtft/LCD-show.git

BillyBlaze, 2020-05-18:
ok, I think the file `/usr/share/X11/xorg.conf.d/99-fbdev.conf` isn't created by the driver, and therfor TouchUI is not copying the config into that file.

Can you try this:
1. `sudo touch /usr/share/X11/xorg.conf.d/99-fbdev.conf`
2. Re-run the install script

Pickass1988, 2020-05-18:
Didnt help unfortunately. 

My screen driver gives me the following message when i install it. Might help. 

dpkg: dependency problems prevent removal of xserver-xorg-input-evdev:
 xserver-xorg depends on xserver-xorg-input-all | xorg-driver-input; however:
  Package xserver-xorg-input-all is not installed.
  Package xorg-driver-input is not installed.
  Package xserver-xorg-input-evdev which provides xorg-driver-input is to be removed.

dpkg: error processing package xserver-xorg-input-evdev (--purge):
 dependency problems - not removing
Errors were encountered while processing:
 xserver-xorg-input-evdev
2020
need to update touch configuration
Selecting previously unselected package xserver-xorg-input-evdev.
(Reading database ... 55602 files and directories currently installed.)
Preparing to unpack .../xserver-xorg-input-evdev_1%3a2.10.6-1+b1_armhf.deb ...
Unpacking xserver-xorg-input-evdev (1:2.10.6-1+b1) over (1:2.10.6-1+b1) ...
Setting up xserver-xorg-input-evdev (1:2.10.6-1+b1) ...
Processing triggers for man-db (2.8.5-2) ...

ChristopherRush, 2020-05-18:
You could try to install OctoPrint on the latest Raspbian and then install
the touchscreen. I had to do it this way and it worked fine as the
pre-compiled image was not.

Pickass1988, 2020-05-18:
Yeah i know a workaround - when i install the dekstop everything works fine

BillyBlaze, 2020-05-18:
I wouldn't install the desktop mode; but use OctoPi or the latest Raspbian. Installing desktop mode will clutter your device with unnecessary libs and configs. But if it works for you; go ahead :)

As for fixing this issue; I dont want to burn my finger to figure out what is going wrong in their image.

Pickass1988, 2020-05-18:
Okay thank you anyways :-) I'll try the Raspbian with Octorpint

Pickass1988, 2020-05-18:
I've set this up and it works - but without the webcam  -  the webcam works on the browser, but not on touch ui - is this a missing codec issue? The snapshot works though.

Pickass1988, 2020-05-18:
nvmd. i had to put the ip in front of the /webcam/?action=stream

--> http://192.168.178.70/webcam/?action=stream

closed 2020-06-09 by stale[bot]