BillyBlaze/OctoPrint-TouchUI #405: Installed TouchUI, On Display the Linux-Promt is shown, no TouchUI

freak1051, 2020-05-23:
Today i Installed TouchUI in my Raspberry 4. 

i go throu a Walktrouhg on the Internet. The all are mostly the same. i Have a 5" 800 x 480 Display with XPT2046 Touchdriver. 

After following the steps (Download and install Chromium, run the TouchUI install routine, set an Autologin User, Enable Autologin) after reboot the Display show the Linux-Promt (" pi@octopi:~ "

what am i doing wrong?

AtomicRas, 2020-05-24:
Having the same issue, i'm using the ELECROW 5 Inch Touch Screen 800x480

https://www.amazon.com/gp/product/B013JECYF2/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

chrisnurse, 2020-06-03:
My issue may be similar, I installed lightdm to get my display running, then I ran through the TouchUI install and boot to browser. But my pi boots to login prompt.

When asked for the user that can autologin, is that the linux user or the OctoPrint user?

closed 2020-06-26 by stale[bot]