BillyBlaze/OctoPrint-TouchUI #407: touchscreen black until touched.

Blaizefed, 2020-06-04:
PI 3b+ running octoprint. with a 3.5 inch SPI touchscreen from UCTRONICS.

boots and loads fine. and the screen works fine with the native Raspbian desktop. Touch all works as it should, however on boot up with TouchUI I get the TouchUI logo for 5-10 seconds or so followed by a blank/black screen with a (non flashing) cursour in the top left corner. If I press the screen with my finger/stylus I can see the TouchUI GUI in there end even interact with it (open the menu, scroll up and down, etc.). But as soon as I release pressure it goes back to black. octoprint is working fine in the background, I can log into it with PC and have the normal interface. When the shutdown command is given via PC, the 3.5" screen dumps TouchUI and reverts back to command line where again it is working fine. 

I don't really know what I am doing here and have been only just treading water up to this point. totally stumped as to what could be causing this issue.

NiicOw974, 2020-06-19:
I have exactly the same problem with the same configuration.

NiicOw974, 2020-06-19:
I resolve the problem... Use LCD-Show driver rather than Elecrow-LCD35

[https://github.com/goodtft/LCD-show](url)

sudo LCD-show/LCD35-show

closed 2020-07-11 by stale[bot]