BillyBlaze/OctoPrint-TouchUI #413: Misplaced buttons and a missing file list on 1.4.1rc3

cp2004, 2020-07-16:
Just installed RC3 for OctoPrint 1.4.1, and TouchUI is missing the file lists on the , as well as the control tab looking messy with the new buttons. Here's a screenshot:
[image]

File list is missing from the #control tab too. Let me know if you need more info

cp2004, 2020-07-18:
I also had the issue from https://github.com/OctoPrint/OctoPrint/issues/3646 where I couldn't get out of TouchUI (the settings panel disappeared). I also found that the system commands were missing from the menu too. Not sure what the cause of this could be, but happens on multiple devices.

foosel, 2020-07-21:
FYI, planned release (unless anything else comes up) would be early next week, so July 27th/28th-ish.

foosel, 2020-07-21:
Issue appears to be caused by the bootstrap JS making some wrong assumptions about the HTML structure now...

Replacing

``` js
TouchUI.prototype.DOM.move.controls = {

	init: function() {

		// backward compatibility with <1.3.0
		if($('#control-jog-feedrate').length === 0) {
			var jogPanels = $('#control > .jog-panel');

			$(jogPanels[0]).find(".jog-panel:nth-child(1)").attr("id", "control-jog-xy");
			$(jogPanels[0]).find(".jog-panel:nth-child(2)").attr("id", "control-jog-z");
			$(jogPanels[1]).attr("id", "control-jog-extrusion");
			$(jogPanels[2]).attr("id", "control-jog-general");

			$('<div class="jog-panel" id="control-jog-feedrate"></div>').insertAfter($(jogPanels[2]));
			$(jogPanels[0]).find("> button:last-child").prependTo("#control-jog-feedrate");
			$(jogPanels[0]).find("> input:last-child").prependTo("#control-jog-feedrate");
			$(jogPanels[0]).find("> .slider:last-child").prependTo("#control-jog-feedrate");

		}

		$("#control-jog-feedrate").insertBefore("#control-jog-extrusion");
		$("#control-jog-extrusion button:last-child").prependTo("#control-jog-feedrate");
		$("#control-jog-extrusion input:last-child").attr('data-bind', $("#control-jog-extrusion input:last-child").attr('data-bind').replace('slider: {', 'slider: {tools: tools(), ')).prependTo("#control-jog-feedrate");
		$("#control-jog-extrusion .slider:last-child").prependTo("#control-jog-feedrate");
		$("#control div.distance").prependTo("#control-jog-feedrate");
	}

}
```

with

``` js
TouchUI.prototype.DOM.move.controls = {

	init: function() {

		// backward compatibility with <1.3.0
		if($('#control-jog-feedrate').length === 0) {
			var jogPanels = $('#control > .jog-panel');

			$(jogPanels[0]).find(".jog-panel:nth-child(1)").attr("id", "control-jog-xy");
			$(jogPanels[0]).find(".jog-panel:nth-child(2)").attr("id", "control-jog-z");
			$(jogPanels[1]).attr("id", "control-jog-extrusion");
			$(jogPanels[2]).attr("id", "control-jog-general");

			$('<div class="jog-panel" id="control-jog-feedrate"></div>').insertAfter($(jogPanels[2]));
			$(jogPanels[0]).find("> button:last-child").prependTo("#control-jog-feedrate");
			$(jogPanels[0]).find("> input:last-child").prependTo("#control-jog-feedrate");
			$(jogPanels[0]).find("> .slider:last-child").prependTo("#control-jog-feedrate");

		}

		if($('#control-jog-feedrate .input-append').length === 0) {
			// <1.4.1
			$("#control-jog-feedrate").insertBefore("#control-jog-extrusion");
			$("#control-jog-extrusion button:last-child").prependTo("#control-jog-feedrate");
			$("#control-jog-extrusion input:last-child").attr('data-bind', $("#control-jog-extrusion input:last-child").attr('data-bind').replace('slider: {', 'slider: {tools: tools(), ')).prependTo("#control-jog-feedrate");
			$("#control-jog-extrusion .slider:last-child").prependTo("#control-jog-feedrate");
		} else {
			// >=1.4.1
			$("#control-jog-feedrate").insertBefore("#control-jog-extrusion");
			$("#control-jog-feedrate .input-append button").insertAfter("#control-jog-feedrate .input-append");

			$("#control-jog-flowrate").removeClass("jog-panel");
			$("#control-jog-flowrate .input-append button").insertAfter("#control-jog-flowrate .input-append");
		}

		$("#control div.distance").prependTo("#control-jog-feedrate");
	}

}
```

in `touchui.bundled.js` makes the UI functional again.

Trying to figure out what that gets generated from...

cp2004, 2020-07-21:
Cool, will check that out later. Should have probably looked at the error console, but didn't have time when I reported this (and since forgot....) Will let you know if there's any other issues

cp2004, 2020-08-04:
@BillyBlaze Any idea when this will be released? OctoPrint 1.4.1 is out today

spanzetta, 2020-08-04:
I confirm these big problems after updated to 1.4.1..
1) question mark as menu
2) completelly wrong interface in almost all tabs
3) once enabled.. I found no way to disable Touch UI from interface
I hope it will be fixed soon..
In this state... it is unusable

cp2004, 2020-08-04:
@spanzetta If you can get into the plugin manager (You may have to start in safe mode from a new browser) You can try installing the fixed version in maintenance 
Enter this url in the plugin manager under Get more > ...from url

`https://github.com/BillyBlaze/OctoPrint-TouchUI/archive/maintenance.zip`

Hopefully that fixes it

spanzetta, 2020-08-04:
> @spanzetta If you can get into the plugin manager (You may have to start in safe mode from a new browser) You can try installing the fixed version in maintenance
> Enter this url in the plugin manager [...]

Thanks for you very quick reply..
As soon as I finish the current print (cannot manage plugins while printing) I will do what you suggest
Thanks again

moof-src, 2020-08-04:
The fix appears to partially work running Chrome on Mac OS Catalina
10.15.5.  When TouchUI is installed, the initial browser regardless of
size (full screen 4k) starts in TouchUI mode.  The menus are working
again and I can disable TouchUI on the desktop.  It also works on an
iPhone 7.

Thank you.

cp2004, 2020-08-04:
I take no credit for the fix, just providing the workaround!

spanzetta, 2020-08-04:
I confirm it fixed many problems.. but not all..
There are still some issues with some other plugins/menu that appears as ?
Also.. pls note that there is an issue with Octolapse plugin (the graphics content covers the menu list)

BillyBlaze, 2020-08-04:
This fix has been released in latest version (0.3.15)

PR #414 (2020-07-21): Adapt control moving to OctoPrint 1.4.1
Feedrate/flowrate controls have been changed, making the old implementation
fail with a JS error. This works around that.

Note: 1.4.1rc3 still has a duplicated id here (`control-jog-feedrate` exists twice,
second one should be `control-jog-flowrate`), I'll change that for the stable final
release.

Closes #413
closed 2020-08-04 by BillyBlaze