BillyBlaze/OctoPrint-TouchUI #417: TouchUI Not Showing Fullscreen on Raspberry Pi Display

rudyamid, 2020-08-04:
I'm using the Raspberry Pi 7" touch display with a Pi4 running the latest Octoprint update 1.4.1.  The version of TouchUI plugin is 0.3.14.  Before the update, the browser would fill the entire fullscreen of the dispaly.  But after the update, it will only show slightly more than half the size.  Attached is the photo of what the screen looks like.  In the past I had to fiddle with the CSS file to force the UI to use fullscreen.  That was fixed with the latest version of TouchUI - now it looks like the problem came back.

Any ideas?

Thanks
[image]

ebrady1, 2020-08-05:
I have this issue also after updating to Octoprint 1.4.1 and using a 7" touchscreen.

ebrady1, 2020-08-06:
Updating to 0.3.15 appears to have corrected most of the sizing problems for me.

rudyamid, 2020-08-06:
Sure, it fixed the fullscreen window, but for the shutdown screen, the "Proceed" red button is stuck beyond the bottom of the screen again. 

[image]

ebrady1, 2020-08-07:
Yeah, I just noticed this.    I see the same behavior when I cancel a print from touchscreen (no buttons).  The weird thing is that each time, I would go to my PC and cancel the print, and when the print would cancel, the buttons would appear on the cancel dialog on the touchscreen, but only after the print cancelled.  I repeated this several times.

gms4175, 2020-08-07:
I also see this effect.  Shuts down fine from a PC screen - I just can't do it from the RPi 7" screen.  At all the latet versions.

Pseudo-Nym, 2020-08-10:
Same on 3.5" screen with content in modal dialogs being off screen and no way to scroll down to the button. Also note that trying to scroll to get to selections further down the main drop down menu is sketchy at best.

aaronvogel, 2020-08-13:
Just came in to mention the same issues of missing buttons in the confirmation dialogs. I just did a fresh install of Octoprint and TouchUI for my new machine. Running on a 3B+ with the Raspberry Pi official 7" display. It isn't just on the system shut down dialog but on all the "are you sure" type dialogs.

blankink65, 2020-08-14:
Seeing the same on Raspberry Pi 7 inch touch screen and Android Chrome browser. All confirmation dialog windows with the "are you sure" cannot see or access the cancel or proceed options. I also enabled Touch-ui on Windows 10 in Chrome browser, same issue present.

aaronvogel, 2020-08-14:
I just enabled TouchUI on my desktop browsers - Firefox and Safari - and the buttons are indeed there, they just are all the way at the bottom. The dialog scales to the whole browser size if changed.

If scaled to the point where it can't any more then the window scrolling does work.

[image]
[image]

blankink65, 2020-08-14:
I tried the same in Windows with Chrome browser, unable to see or access options.
[image]

TheNore, 2020-08-14:
this is happening to me as well on 2x rpi 7" instances

blankink65, 2020-08-14:
There is mention in another issue of editing a CSS file as a workaround. Not sure about the process, I think I will just wait for an update.

rudyamid, 2020-08-29:
Any Octoprint or TouchUI developer looking into this?  The case can't be closed when there's no resolution.

Pseudo-Nym, 2020-08-29:
My off the cuff hypothesis would be it has something to do with the overflow: hidden in the modal.less or possibly where the buttons reside in the DOM of the modals being generated.

CTcartesian3D, 2020-08-30:
https://community.octoprint.org/t/touchui-temperature-graph-squished-modify-size/3014/21

This may help and give you an idea where to start...

BillyBlaze, 2020-09-09:
Thanks for reporting; closing this issue because there is already an open issue @ #418

closed 2020-09-09 by BillyBlaze