BillyBlaze/OctoPrint-TouchUI #425: Scrollbar wont go away and dialogs too tall

lciscon, 2020-08-17:
I just recently upgraded to TouchUI 0.3.15/Octoprint 1.4.2 and have encountered two new issues:

First, on any page that doesn't fit vertically, the vertical scrollbar stays on the screen now and doesn't go away.  In the past it would disappear once you stopped scrolling.  Is there anyway to turn it off or hide it?

Second,  when I go to cancel a print job, the dialog box is too big and the buttons are off the screen. See attached picture. I actually think this is true with all of the pop-up dialogs.

I'm using a 1024x600 USB touch display.

Any suggestions?

asmagill, 2020-08-19:
I'll also confirm the scrollbar issue. And it appeared as soon as I upgraded Octoprint to 1.4.2.

As to the hidden buttons for the pop-ups, a temporary fix has been suggested at https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/418#issuecomment-671082227, and it seems to work for me, at least for things like restarting Ocotprint and updating plugins (just saw that today and was glad I didn't have to go to the other room and log on from my desktop).

blankink65, 2020-08-21:
I can confirm that the temporary fix from issue #418 fixes the buttons on the confirmation dialog.

andysmithfal, 2020-08-21:
Edit `/home/pi/oprint/lib/python2.7/site-packages/octoprint_touchui/static/css/touchui.css` and add this at the top to get rid of the scrollbar: 

```
::-webkit-scrollbar {
    display: none;
}
```

cannfoddr, 2020-09-01:
That file for me is one massive length single line - I am not even sure if vi is capable of editing it!

BillyBlaze, 2020-09-09:
Thanks for reporting; closing this issue because there is already an open issue @ #418

closed 2020-09-09 by BillyBlaze