BillyBlaze/OctoPrint-TouchUI #427: 0.3.15 Dialog Buttons not visible on touchscreen and smartphone

MarioF84, 2020-08-28:
I recently set up Octopi with TouchUI and I am encountering the problem, that dialogs for confirmations and settings do not fit on the screen (only height) - so the buttons are always invisible, sometimes (in the settings dialog) even some part of the content is missing.
Looks exactly like the already reported issues [296](https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/296), [377](https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/377) and [384](https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/384).
I am running OctoPrint 1.4.2 with TouchUI 0.3.15 (Touchscreen has been set up according to [Setup: Boot to Browser](https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Boot-to-Browser))

Octoprint is running on a Pi 4 (2GB), with the official 7" Touchscreen.

The problem occurs on the touchscreen as well as on my phone (Wiko View 3 Pro). On the phone it doesn't matter if in landscape or portrait mode or if fullscreen or not.
When I enable TouchUI in my Desktop Browser it works fine, even if I use Chromes developer tools to simulate different phone screens.

As long as I am not in any dialog, I am able to scroll down completely and see the footer.

cannfoddr, 2020-09-01:
Everything was fine with octoprint until a the most recent updates.  Not sure what changed but my 7" official touchscreen pi no longer has buttons visible on key dialogs. eg. cancel print.  I simply cannot cancel a print from touch on the screen anymore.  I have tried upgrading to the experimental version of touch mentioned in some related issues - this has not helped.

I have seen suggestions to tweak the css file - mine (and the one on GitHub) is a massive single line file that I am not comfortable editing for fear of messing thins up completely (why is is formatted like this?)

Really want working dialogs back?

gperrone71, 2020-09-07:
Same problem here unfortunately: running latest OctoPi with TouchUI on a 7' display on a 1024 x 600 res and the buttons on the dialog are below the screen. Interestingly enough, I have noticed that swiping left (and therefore changing the page in the background) sometimes make appear the upper part of the dialogs enough so that I can push them. 

Very unfortunate though since I like TouchUI a lot and not being able to use dialogs complicate things a lot!

cannfoddr, 2020-09-07:
I seem to have resolved my main issues (possibly all of them) by following the steps outlined here:
 
https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/418#issuecomment-677983625

and using the latest css in that thread.

gperrone71, 2020-09-08:
@cannfoddr Thanks! Took some tweaking with vi but it eventually worked!!!

BillyBlaze, 2020-09-09:
Thanks for reporting; closing this issue because there is already an open issue @ #418

closed 2020-09-09 by BillyBlaze