BillyBlaze/OctoPrint-TouchUI #428: Help

VASAPOL, 2020-08-29:
I use pi 3 b+ i install octoprint after i install it work properly and i install TouchUI and auto start my octoprint shutdown
logs:
2020-08-29 04:13:59,553 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2020-08-29 04:13:59,575 - octoprint.server - INFO - --- Log roll over detected ---------------------------------------------------
2020-08-29 04:13:59,576 - octoprint.server - INFO - OctoPrint 1.3.12
2020-08-29 04:13:59,582 - octoprint.plugin.core - INFO - 15 plugin(s) registered with the system:
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/appkeys
|  Backup & Restore (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/backup
|  Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/corewizard
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/discovery
|  Error Tracking (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/errortracking
|  Force Login (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/forcelogin
|  Logging (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/logging
|  Pi Support Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pi_support
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager
|  Printer Safety Check (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/printer_safety_check
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/virtual_printer
2020-08-29 04:13:59,609 - octoprint.environment - INFO - Detected environment is Python 2.7.16 under Linux (linux2). Details:
|  hardware:
|    cores: 4
|    freq: 1400.0
|    ram: 917016576
|  os:
|    id: linux
|    platform: linux2
|  plugins:
|    pi_support:
|      model: Raspberry Pi 3 Model B Plus Rev 1.3
|      octopi_version: 0.17.0
|  python:
|    pip: 19.3.1
|    version: 2.7.16
|    virtualenv: /home/pi/oprint
2020-08-29 04:13:59,610 - octoprint.server - INFO - ------------------------------------------------------------------------------
2020-08-29 04:13:59,795 - octoprint.plugins.announcements - INFO - Loaded channel _releases from https://octoprint.org/feeds/releases.xml in 0.79s
2020-08-29 04:14:00,897 - octoprint.plugins.announcements - INFO - Loaded channel _blog from https://octoprint.org/feeds/octoblog.xml in 0.9s
2020-08-29 04:14:01,847 - octoprint.plugins.announcements - INFO - Loaded channel _plugins from https://plugins.octoprint.org/feed.xml in 0.8s
2020-08-29 04:14:02,922 - octoprint.plugins.announcements - INFO - Loaded channel _octopi from https://octoprint.org/feeds/octopi.xml in 0.85s
2020-08-29 04:14:56,312 - octoprint.server.util.flask.PreemptiveCache - INFO - Adding entry for / and {'query_string': 'l10n=en', 'path': u'/', '_count': 1, '_timestamp': 1598670896.312128, 'base_url': u'http://192.168.2.176/'}
2020-08-29 04:15:06,105 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.2.38
2020-08-29 04:15:26,314 - octoprint.server.api - INFO - Actively logging in user Maker from ::ffff:192.168.2.38
2020-08-29 04:15:27,558 - octoprint.plugins.announcements - INFO - Loaded channel _important from https://octoprint.org/feeds/important.xml in 0.76s
2020-08-29 04:15:28,396 - octoprint.server.api - ERROR - Error calling SimpleApiPlugin pluginmanager
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/__init__.py", line 68, in pluginData
    response = api_plugin.on_api_get(request)
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 280, in on_api_get
    if refresh_repository or not self._is_repository_cache_valid():
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 797, in _is_repository_cache_valid
    return mtime + self._repository_cache_ttl >= time.time() > mtime
TypeError: unsupported operand type(s) for +: 'NoneType' and 'int'
2020-08-29 04:15:28,402 - tornado.access - ERROR - 500 GET /api/plugin/pluginmanager (::ffff:192.168.2.38) 12.89ms
2020-08-29 04:15:30,504 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python2 -m pip" as command to invoke pip
2020-08-29 04:15:36,924 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python2.7/site-packages/ (writable -> yes), --user flag needed -> no, virtual env -> yes
2020-08-29 04:15:36,925 - octoprint.util.pip - INFO - ==> pip ok -> yes
2020-08-29 04:15:36,958 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2020-08-29 04:15:37,685 - octoprint.server.util.flask - INFO - Passively logging in user Maker from ::ffff:192.168.2.38
2020-08-29 04:15:43,399 - octoprint.server.util.sockjs - INFO - User Maker logged in on the socket from client ::ffff:192.168.2.38
2020-08-29 04:15:43,432 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.2.38
2020-08-29 04:15:43,697 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.2.38
2020-08-29 04:15:43,786 - octoprint.server.util.flask - INFO - Passively logging in user Maker from ::ffff:192.168.2.38
2020-08-29 04:15:45,808 - octoprint.server.api - ERROR - Error calling SimpleApiPlugin pluginmanager
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/__init__.py", line 68, in pluginData
    response = api_plugin.on_api_get(request)
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 280, in on_api_get
[... 302 lines cut ...]
2020-08-29 04:20:51,613 - octoprint.server.util.flask - INFO - Passively logging in user Maker from ::ffff:192.168.2.38
2020-08-29 04:20:51,645 - octoprint.server.util.flask - INFO - Passively logging in user Maker from ::ffff:192.168.2.38
2020-08-29 04:20:51,671 - tornado.access - WARNING - 404 GET /static/webassets/packed_libs.css?ec88168c (::ffff:192.168.2.38) 14.86ms
2020-08-29 04:20:51,680 - tornado.access - WARNING - 404 GET /static/webassets/packed_core.css?0635c0de (::ffff:192.168.2.38) 4.50ms
2020-08-29 04:20:51,691 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.2.38
2020-08-29 04:20:51,694 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.2.38
2020-08-29 04:20:51,711 - tornado.access - WARNING - 404 GET /static/webassets/packed_libs.js?ef1e2987 (::ffff:192.168.2.38) 3.56ms
2020-08-29 04:20:51,722 - tornado.access - WARNING - 404 GET /static/webassets/packed_client.js?68a8d0b0 (::ffff:192.168.2.38) 7.06ms
2020-08-29 04:20:51,731 - tornado.access - WARNING - 404 GET /static/webassets/packed_plugins.js?b63d97ee (::ffff:192.168.2.38) 5.08ms
2020-08-29 04:20:51,738 - tornado.access - WARNING - 404 GET /static/webassets/packed_core.js?b00f909b (::ffff:192.168.2.38) 2.50ms
2020-08-29 04:21:13,440 - octoprint.server.util.flask - INFO - Passively logging in user Maker from ::ffff:192.168.2.38
2020-08-29 04:21:13,737 - tornado.access - WARNING - 404 GET /static/webassets/packed_core.js?b00f909b (::ffff:192.168.2.38) 9.02ms
2020-08-29 04:21:13,771 - tornado.access - WARNING - 404 GET /static/webassets/packed_core.js?b00f909b (::ffff:192.168.2.38) 7.14ms
2020-08-29 04:21:15,225 - octoprint.server.preemptive_cache - INFO - ... done in 16.56s
2020-08-29 04:21:24,961 - octoprint.server.util.flask - INFO - Passively logging in user Maker from ::ffff:192.168.2.38
2020-08-29 04:21:25,305 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.2.38
2020-08-29 04:21:25,360 - octoprint.server.util.flask - INFO - Passively logging in user Maker from ::ffff:192.168.2.38
2020-08-29 04:21:26,520 - octoprint.server.util.sockjs - INFO - User Maker logged in on the socket from client ::ffff:192.168.2.38
[... cut ...]
2020-08-29 04:28:22,351 - octoprint.server.util.sockjs - INFO - Client connection closed: ::1
2020-08-29 04:28:23,861 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.2.38
2020-08-29 04:28:23,959 - octoprint.server - INFO - Shutting down...
2020-08-29 04:28:24,503 - octoprint.events - INFO - Processing shutdown event, this will be our last event
2020-08-29 04:28:24,505 - octoprint.events - INFO - Event loop shut down
2020-08-29 04:28:24,543 - octoprint.server - INFO - Goodbye!

VASAPOL, 2020-08-29:
@BillyBlaze Please Help me

devildant, 2020-09-11:
I think your problem is:
2020-08-29 04:13:59,576 - octoprint.server - INFO - OctoPrint **1.3.12**

the latest version of octoprint is as follows :  **1.4.2**

closed 2020-12-08 by github-actions[bot]

thijsdeschildre, 2020-12-22:
I'm having the same issue. Octoprint  1.5.2, TouchUI 0.3.17.
I get to the login screen, but can't log in. It just reloads the login page. Using the wrong password, I get the "wrong password" error. log files show how the connection immediately gets closed:

020-12-22 12:47:48,591 - octoprint.access.users - INFO - Logged in user: teus
2020-12-22 12:47:48,597 - octoprint.server.api - INFO - Actively logging in user teus from ::1
2020-12-22 12:47:48,811 - octoprint.server.util.sockjs - INFO - Client connection closed: ::1
2020-12-22 12:47:49,068 - octoprint.server.util.sockjs - INFO - New connection from client: ::1
2020-12-22 12:48:03,344 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.0.151

thijsdeschildre, 2020-12-22:
I changed the logging for touchui to debug:

2020-12-22 12:55:05,996 - octoprint.access.users - INFO - Logged in user: teus
2020-12-22 12:55:06,003 - octoprint.server.api - INFO - Actively logging in user teus from ::1
2020-12-22 12:55:06,033 - octoprint.events.fire - DEBUG - Firing event: UserLoggedIn (Payload: {u'username': 'teus'})
2020-12-22 12:55:06,047 - octoprint.events - DEBUG - Sending action to <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x695bb050>>
2020-12-22 12:55:06,064 - octoprint.plugin - DEBUG - Calling on_event on action_command_notification
2020-12-22 12:55:06,066 - octoprint.plugin - DEBUG - Calling on_event on action_command_prompt
2020-12-22 12:55:06,073 - octoprint.plugin - DEBUG - Calling on_event on announcements
2020-12-22 12:55:06,075 - octoprint.plugin - DEBUG - Calling on_event on file_check
2020-12-22 12:55:06,077 - octoprint.plugin - DEBUG - Calling on_event on firmware_check
2020-12-22 12:55:06,084 - octoprint.plugin - DEBUG - Calling on_event on pluginmanager
2020-12-22 12:55:06,087 - octoprint.plugin - DEBUG - Calling on_event on softwareupdate
2020-12-22 12:55:06,090 - octoprint.plugin - DEBUG - Calling on_event on tracking
2020-12-22 12:55:06,093 - octoprint.plugin - DEBUG - Calling on_event on bedlevelvisualizer
2020-12-22 12:55:06,119 - octoprint.server.util.flask - DEBUG - Checking if remote address ::1 is in localNetworks (IPSet(['127.0.0.0/8', '::1/128', '::ffff:127.0.0.0/104']))
2020-12-22 12:55:06,120 - octoprint.plugin - DEBUG - Calling on_event on firmwareupdater
2020-12-22 12:55:06,137 - octoprint.plugin - DEBUG - Calling on_event on thespaghettidetective
2020-12-22 12:55:06,146 - octoprint.plugin - DEBUG - Calling on_event on youtubelive
2020-12-22 12:55:06,148 - octoprint.events.fire - DEBUG - Firing event: UserLoggedIn (Payload: {u'username': None})
2020-12-22 12:55:06,149 - octoprint.events - DEBUG - Sending action to <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x695bb050>>
2020-12-22 12:55:06,160 - octoprint.plugin - DEBUG - Calling on_event on action_command_notification
2020-12-22 12:55:06,162 - octoprint.plugin - DEBUG - Calling on_event on action_command_prompt
2020-12-22 12:55:06,164 - octoprint.plugin - DEBUG - Calling on_event on announcements
2020-12-22 12:55:06,166 - octoprint.plugin - DEBUG - Calling on_event on file_check
2020-12-22 12:55:06,169 - octoprint.plugin - DEBUG - Calling on_event on firmware_check
2020-12-22 12:55:06,172 - octoprint.plugin - DEBUG - Calling on_event on pluginmanager
2020-12-22 12:55:06,184 - octoprint.server.util.flask - DEBUG - Checking if remote address ::1 is in localNetworks (IPSet(['127.0.0.0/8', '::1/128', '::ffff:127.0.0.0/104']))
2020-12-22 12:55:06,185 - octoprint.plugin - DEBUG - Calling on_event on softwareupdate
2020-12-22 12:55:06,188 - octoprint.plugin - DEBUG - Calling on_event on tracking
2020-12-22 12:55:06,191 - octoprint.plugin - DEBUG - Calling on_event on bedlevelvisualizer
2020-12-22 12:55:06,195 - octoprint.plugin - DEBUG - Calling on_event on firmwareupdater
2020-12-22 12:55:06,209 - octoprint.plugin - DEBUG - Calling on_event on thespaghettidetective
2020-12-22 12:55:06,212 - octoprint.plugin - DEBUG - Calling on_event on youtubelive
2020-12-22 12:55:06,219 - octoprint.events.fire - DEBUG - Firing event: UserLoggedIn (Payload: {u'username': None})
2020-12-22 12:55:06,220 - octoprint.events - DEBUG - Sending action to <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x695bb050>>
2020-12-22 12:55:06,232 - octoprint.plugin - DEBUG - Calling on_event on action_command_notification
2020-12-22 12:55:06,233 - octoprint.plugin - DEBUG - Calling on_event on action_command_prompt
2020-12-22 12:55:06,234 - octoprint.plugin - DEBUG - Calling on_event on announcements
2020-12-22 12:55:06,235 - octoprint.plugin - DEBUG - Calling on_event on file_check
2020-12-22 12:55:06,238 - octoprint.plugin - DEBUG - Calling on_event on firmware_check
2020-12-22 12:55:06,241 - octoprint.plugin - DEBUG - Calling on_event on pluginmanager
2020-12-22 12:55:06,250 - octoprint.plugin - DEBUG - Calling on_event on softwareupdate
2020-12-22 12:55:06,256 - octoprint.plugin - DEBUG - Calling on_event on tracking
2020-12-22 12:55:06,258 - octoprint.plugin - DEBUG - Calling on_event on bedlevelvisualizer
2020-12-22 12:55:06,261 - octoprint.plugin - DEBUG - Calling on_event on firmwareupdater
2020-12-22 12:55:06,266 - octoprint.plugin - DEBUG - Calling on_event on thespaghettidetective
2020-12-22 12:55:06,272 - octoprint.plugin - DEBUG - Calling on_event on youtubelive
2020-12-22 12:55:06,413 - octoprint.events.fire - DEBUG - Firing event: ClientClosed (Payload: {u'remoteAddress': u'::1'})
2020-12-22 12:55:06,415 - octoprint.events - DEBUG - Sending action to <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x695bb050>>
2020-12-22 12:55:06,414 - octoprint.server.util.sockjs - INFO - Client connection closed: ::1