BillyBlaze/OctoPrint-TouchUI #433: Boot to browser cannot get to TouchUI

dchauran, 2020-09-17:
When I boot to TouchUI, octoprint does not start touchui, even after setting FORCE_TOUCH=true in /etc/defaults/touchui/. Moreover, the option to enter touchui is not there.

I can connect with a browser from a PC and switch to TouchUI as well as load touchui on my phone.

Pi 4, Octopi image. 7" touchscreen connected via HDMI1 & USB. Native 1024x600, running 1080p, but have also tried 1024x768.

I have updated everything to latest:

TouchUI (0.3.16)
Linux octopi 5.4.65-v7l+ #1341 SMP Tue Sep 15 13:13:32 BST 2020 armv7l GNU/Linux
octoprint, version 1.4.2
TouchUI-autostart commit a90b4deffad3bb879ecc8155eb1df6d94d7b66ec
chromium-browser/testing,now 78.0.3904.108-rpt1 armhf

dchauran, 2020-09-18:
This appears to be a problem with chromium-browser vs chromium. apt install chromium removes the chromium-browser package and install just chromium. This works fine when I access from a desktop session, but does not work with the touchui autostart. I had to change the URL to just got to localhost/#touch with no port, and lose the fancy loading screen.

GitFreed, 2020-11-06:
Hello, I tried what you did (force touch TRUE and `TOUCHUI_URL="http://localhost:$TOUCHUI/"` ?) but it's not working, do I need chromium only?

I get an error : 

```
pi@8freed:~ $ apt install chromium
E: Could not open lock file /var/lib/dpkg/lock-frontend - open (13: Permission denied)
E: Unable to acquire the dpkg frontend lock (/var/lib/dpkg/lock-frontend)
```

GitFreed, 2020-11-06:
Hello, I tried what you did, installed chromium and changed in /etc/default/touchui : Force touch TRUE and `TOUCHUI_URL="http://localhost:$TOUCHUI/"` (?) but it's not working I got this error :

```
pi@8freed:~ $ chromium
[1161:1161:1106/002105.347334:ERROR:browser_main_loop.cc(1473)] Unable to open X display.
```

DanielYWoo, 2020-11-06:
@GitFreed same here. I can startx, but cannot launch chromium, exactly the same error as you.

closed 2020-12-08 by github-actions[bot]