BillyBlaze/OctoPrint-TouchUI #435: error with touchUI helpers install for booting to browser

armon4, 2020-09-21:
when I try to do the final step for booting to browser I keep getting this error.

[image]

Will-wastelander, 2020-10-14:
> when I try to do the final step for booting to browser I keep getting this error.
> 
> [image]

What happens if you run apt upgrade manually ?

AP4130, 2020-11-23:
What version of Octopi are you running? I got the same error with 0.18.0rc1; got a little further by installing the Desktop environment, but still failed in the end.  I fixed the problem by reverting back to Octopi 0.17.0.

closed 2020-12-15 by github-actions[bot]