BillyBlaze/OctoPrint-TouchUI #439: Unable to connect when running Octoprint 1.4.2

Will-wastelander, 2020-10-13:
I know there was a previous issue reported about the same problem, and it looks like it has been closed. I just did an install of Octopi 0.17.0, and current TouchUI, and I am having the issue of "[::1] refused to connect." I have attempted to use the fix by changing localhost to be rPi's hostname, and this did not work either. I did a reinstall of 0.17.0, which comes with Octoprint 1.3.12, and TouchUI works w/o issues. Has there been any update on this bug ?

If I need to provide additional info to assist with troubleshooting or debugging, please let me know what to provide, and I will work on getting the info.

Will-wastelander, 2020-10-13:
This was the previous bug report.

https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/368

Will-wastelander, 2020-10-13:
I have also updated python from 2.x to 3, but this issue existed before the python upgrade.

Will-wastelander, 2020-10-13:
Oh, and I have disabled IPv6 w/ sysctl.conf using this page's info. https://cwesystems.com/?p=231

Will-wastelander, 2020-10-15:
I found the resolution to this issue.

run the following commands after you have updated to 1.4.2

SSH to Octopi.
`
~/oprint/bin/octoprint config set --bool server.allowFraming true
sudo service octoprint restart
sudo service touchui  restart
`

https://community.octoprint.org/t/my-touchui-display-is-no-longer-working-since-updating-to-octoprint-1-4-0/16819

JMHaro, 2020-12-04:
I just tried the above commands but mine still gives me a "browser's error console octoprint.log" screen.  I still can't get mine started.  This was a new install of everything.

closed 2020-12-11 by github-actions[bot]