BillyBlaze/OctoPrint-TouchUI #441: New install not working

Doormat1, 2020-10-19:
Just tried to install touch ui on my raspberry pi 3.
Screen is working and according to putty touch ui is also installed. However my pi just has the normally load up text and stops at octoprint login with a flashing cursor. Any ideas what I have done wrong? I did update and upgrade before starting.

Doormat1, 2020-10-21:
Just done a fresh install of octoprint, installing 1.3.12 and it works now. Looks like out of the box, this doesn't work with 1.4.3

Krontx, 2020-11-22:
Downgrading to 1.3.6 solves the issue:

Just run:

```
1. ~/oprint/bin/pip install https://github.com/OctoPrint/OctoPrint/archive/1.3.6.zip

2. sudo service octoprint restart
```

JMHaro, 2020-12-04:
I just tried the above commands to downgrade but they didn't work for me.  I'm getting multiple errors like:
 WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x75aca410>: Failed to establish a new connection: [Errno 101] Network is unreachable',)': /OctoPrint/OctoPrint/archive/1.3.6.zip

closed 2020-12-25 by github-actions[bot]