BillyBlaze/OctoPrint-TouchUI #442: Login loop

orangelizard, 2020-11-05:
Login from the pi 3 3.5"LCD results in login screen being presented again instead of heading to the browser interface. It works fine from mobile devices and pc browsers.

GitFreed, 2020-11-05:
Same here, tried every kind of Boot setup linked here but none works, I keep stuck on pi login, then the Octopi terminal "Access Octopi from a web browser etc..." 

It works fine from my PC or smartphone.

Will-wastelander, 2020-11-05:
Did you guys try the fix i had posted in the bug report I posted ?

Will-wastelander, 2020-11-05:
Give this a try..

https://community.octoprint.org/t/my-touchui-display-is-no-longer-working-since-updating-to-octoprint-1-4-0/16819

orangelizard, 2020-11-05:
Firstly thank you for your response. Yes I tried this earlier today and have just repeated it to make sure, sadly the same outcome.

GitFreed, 2020-11-06:
Thank you Will! Yes I just tried and nothing changed. 
It's a fresh install here, not an update problem, but it can also due to this 1.4.2 version, I see lot of people with boot/login problem.
I tried like every setup (old and new) but still cursed on this... asking the pi user/passwd like when I loggin from Putty or SSH terminal... then stuck on the optoprint terminal.

orangelizard, 2020-11-06:
Just to clarify my issue. I installed an LCD on an old updated over the past few years octoprint instance and all works as before with touch ui except I get stuck in a loop at the Touch UI octoprint login screen. So I grabbed a spare card and created a new setup from scratch assuming the old setup was the problem, but still the same. If I enter incorrect login details it let's me know they are incorrect. If I login correctly it just displays the login again. See attached for clarification on where I'm experiencing the login loop. All SSH and external Octoprint with touch ui on external devices work correctly. 
[image]

seanweinert, 2020-11-11:
Same exact issue as orangelizard here, RPi 4 with 3.5 touchscreen... bashing my head against a wall trying to figure this out

orangelizard, 2020-11-11:
It's infuriating, I've temporarily popped Octodash on another card until I have some divine intervention as I've exhausted all possibilities with my limited knowledge. Would definitely prefer touch ui though.

seanweinert, 2020-11-11:
> It's infuriating, I've temporarily popped Octodash on another card until I have some divine intervention as I've exhausted all possibilities with my limited knowledge. Would definitely prefer touch ui though. I'm convinced I'm just doing something totally stupid messing it up.

I've been toying with the idea that this is a chromium problem?

orangelizard, 2020-11-11:
Before trying to add a screen to the pi I used to just use an old HTC phone fixed to the printer with an old version of chrome on and that stopped working being stuck in this same login loop. At the time I thought nothing of it and swapped to a newer phone. This was a while ago though. I hadn't thought about it till now.

seanweinert, 2020-11-11:
like you said earlier - all other current devices load perfectly... which leads to my current thinking - what's up with chromium on this image?

seanweinert, 2020-11-11:
New developments - if you plug in a keyboard and wait for the login screen to load, press cntrl +n, it'll open a new chromium tab. You can navigate to octopi.local or your static IP and login and it DOES log you in. I was even prompted to save password from chrome which i thought may help resolve this problem. once you reboot the pi it defaults back to this login page and no longer works. Unfortunately when you log in on the new tab the site isn't 3.5" optomized and you really can't do much. I'll continue to poke around....

BillyBlaze, 2020-11-12:
We configure the built-in [autoLogin](https://docs.octoprint.org/en/master/configuration/config_yaml.html#access-control) feature inside OctoPrint to prevent the user from seeing this screen when using a touchscreen.

1. Make sure your octoprint config file has the 'autoLoginAs' configured under the section 'accessControl' (if not add them, see the variables over [here](https://docs.octoprint.org/en/master/configuration/config_yaml.html#access-control))
2. Does your octoprint.log contain any errors?

seanweinert, 2020-11-12:
Hi Billy Blaze - I'll look into the log and paste errors here. When i change the autoLoginAs to anything other than "pi" (my username is admin) - it refuses to log in and gives and error.

orangelizard, 2020-11-12:
Thanks Billy Blaze, I also tried with my username which has a ;) at the end of it which I thought was the problem. I'll try again though and find how to check the logs. Also that was the same screen I got trying the auto login as Sean posted.

Jon-So, 2020-11-13:
Just Updated to Python3 and this same thing happened to me... could it be from that?

Jon-So, 2020-11-13:
OK, I know this isn't the exact startup fix we need and someone did mention we could just open a new 'Browser Tab', so I tried that and I was able to login that way. Though, this isn't the fix its still a way to use TouchUI.

1. Hit CTRL + N to open a new Chromium Browser Tab
2. Type the IP address of your Octoprint Connected Printer
3. Login with your credentials.
4. and then do a full screen mode and wollaaaa we have TouchUI again.

Just wanted to post this for people that aren't too tech savvy.
and Yes you probably have to do this each time as its not part of the startup routine, which is a pain.
But its a workaround until this gets figured out.

Jon-So, 2020-11-13:
Also, could it be the way chromium connects to a non-SSL connection that it keeps looping? Can anyone check that?

Krontx, 2020-11-22:
> Just to clarify my issue. I installed an LCD on an old updated over the past few years octoprint instance and all works as before with touch ui except I get stuck in a loop at the Touch UI octoprint [...]

Downgrading to 1.3.6 solves the issue:

Just run:

```
~/oprint/bin/pip install https://github.com/OctoPrint/OctoPrint/archive/1.3.6.zip

```

`sudo service octoprint restart`

Will-wastelander, 2020-11-22:
Guys, this is the same issue that I reported and provided the fix that Gina
provided. Look up touchiu octoprint 1.4.0

Krontx, 2020-11-22:
> Guys, this is the same issue that I reported and provided the fix that Gina provided. Look up touchiu octoprint 1.4.0
> […](#)

Running that fix or any other found in the comments of the active 30 issues didn't do it for me. 

After running

`~/oprint/bin/octoprint config set --bool server.allowFraming true
sudo service octoprint restart
sudo service touchui restart`

I continued to face the same error, TouchUI failed to start on the raspberry pi screen. I did several clean installations on my Pi 4 Model B.

AP4130, 2020-11-22:
Same roadblock for me as Krontx; I have attempted the Allow Framing fix (both via console and the Octopi web UI), updating the bootloader, and rerunning the install script. I've even tried wiping the SD card and doing a fresh install of everything, and playing around with the URL and port settings inside touchui.default. The only thing I haven't done is mucking about with attempting to force IPv4 vs IPv6 as that is well beyond my skill level. 

No luck at all, just load after load sitting with "Connecting to http://[::1]/" at the bottom of the screen. After a long pause, it will generally make it to the "browser's error console" Chromium screen seanweinert made it to. 

This issue is specific to running on the host itself, as TouchUI loads and runs quite nicely through both Edge on my PC and Safari on my iPhone. 

Running OctoPrint 1.4.2, OctoPi 0.17.0, and TouchUI 0.3.16 on a Pi 4B. 
Happy to carry out any troubleshooting steps that will help everyone out (and myself, of course). Thanks in advance!

Krontx, 2020-11-22:
> Same roadblock for me as Krontx; I have attempted the Allow Framing fix (both via console and the Octopi web UI), updating the bootloader, and rerunning the install script. I've even tried wiping [...]

That's exactly what happens to me as well ( All the additional details you just described).

AP4130, 2020-11-24:
@seanweinert, @Krontx, please take a look at issue #445; I think @Craigeous figured it out for us!
@orangelizard, just realized I helped hijack this threat away from your original issue of autologin not working, so sorry!

Krontx, 2020-11-24:
@AP4130, I'm aware @Craigeous is a genius. I commented on the topic as a soon as he came up with the fix, I blew up my Saturday trying to figure it out. Thanks for keeping the community posted. Good luck!

juanssl, 2020-12-09:
For reference in the future...
The problem that I was seeing was similar to the ones described here.
The UI on the raspberry pi screen was failing to load at first and then stuck in loading (blue) forever. I changed the chromium options to be able to debug in the browser and found that the request for `http://:::1/api/settings` was failing so i changed the script under `bootloader/script.js` to use `localhost` regardless of the IPV6 values.
That fixed the problem for me, the fix that AP4130 mentioned from Craigeous was to straight up load the octoprint page bypassing the bootloader which is fine but break a few features that the bootloader provides (I think)
Anyways 😬👍

closed 2020-12-17 by github-actions[bot]