BillyBlaze/OctoPrint-TouchUI #443: Browser error

JeedHome44, 2020-11-09:
Hello,

Sorry, I'm french. So I translated with Google translate:

I just installed a 7 inch screen on my Raspberry Pi 3B.
I followed this tutorial to install TouchUI normally:

http://egamaker.be/octoprint-5-plugin-touchui/

I have the TouchUI loading screen which is displayed on the 7 inch screen with "Connecting to TouchUI" but it makes an "browsers's error console" then, then the red TouchUI screen "Startup failed...".

I see at the bottom of the blue loading page the following link: "Connecting to http://[::1]/
I think the problem is there but I don't know how to solve it.

I specify that TouchUI works correctly on Chrome on my desktop computer. The problem is therefore with the autostart configuration I think.

bougs93, 2020-11-11:
[https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/242]((https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/242))

In case you want to disable it, then add AUTODETECT_PORT=false to your settings under '/etc/default/touchui'

AP4130, 2020-11-24:
Please take a look at issue #445; @Craigeous posted a temp fix to the chromium.xinit file that solved this for me.

closed 2020-12-16 by github-actions[bot]