BillyBlaze/OctoPrint-TouchUI #445: logging error

mramerica36, 2020-11-10:
i've tried to install and log in to touch ui on a raspberry pi 4. the first time i log in it has an error telling me to go to the octoprint.log. after a bit it allows me to retry and it doesnt go any where. ive tried the security override and that seems to have done nothing

jasonstory, 2020-11-18:
The same thing is happening to me.  Saying connecting to HTTP://(::1.)

Craigeous, 2020-11-23:
i was getting this issue as well and i went through a lot of issues in here trying stuff.

here are the steps i used to fix it for me:

`nano ~/TouchUI-autostart/chromium.xinit`

change the following:
```
CHROME_ARGS="   --no-first-run --kiosk $TOUCHUI_ARGS $TOUCHUI_DEBUG
                                --dns-prefetch-disable --disable-sync-preferences --disk-cache-size=1 --disk-cache-dir=/dev/null
                                --disable-java --disable-plugins --disable-extensions --disable-infobars
                                --user-agent='TouchUI (X11, Chrome $CHROME_VERSION$OCTOPI) (P:$PORT)'
                                --start-maximized --window-position=0,0 --user-data-dir=$CONFIG_DIR
                               $TOUCHUI_URL"
```

to this:
```
CHROME_ARGS="   --no-first-run --kiosk $TOUCHUI_ARGS $TOUCHUI_DEBUG
                                --dns-prefetch-disable --disable-sync-preferences --disk-cache-size=1 --disk-cache-dir=/dev/null
                                --disable-java --disable-plugins --disable-extensions --disable-infobars
                                --user-agent='TouchUI (X11, Chrome $CHROME_VERSION$OCTOPI) (P:$PORT)'
                                --start-maximized --window-position=0,0 --user-data-dir=$CONFIG_DIR
                               http://0.0.0.0/"
```

then `sudo service touchui restart` and the ui came right up.

Krontx, 2020-11-23:
Thanks @Craigeous, I will try that after my next print is done tomorrow.

lot-lizard, 2020-11-24:
@Craigeous nice catch.  I didn't even dive in yet and decided to check here first. We obviously want to bring the variable back vs. hardcoded, but gives us an immediate fix and lets @BillyBlaze know where to look.  Im back up as well.  Kudos

AP4130, 2020-11-24:
You got it!!! Thanks so much for this @Craigeous. I blew my entire Sunday trying to get this working, but swapping that one variable out for http://0.0.0.0/ has me up and running. 

Btw, this is also a temp fix for Issue #443, and the tail end of the discussion in issue #442, although reading back up I now realized we kinda hijacked that issue- original poster was having an autologin issue, not the http://::1 problem.

Krontx, 2020-11-25:
Confirmed, up and running after @Craigeous fix. Thanks again!

Craigeous, 2020-11-25:
i'm sure you could just replace localhost in the variable with 0.0.0.0 and it'll work.  i just did it for a quick fix.  using localhost seemed to be more of the issue than anything.

closed 2020-12-17 by github-actions[bot]