BillyBlaze/OctoPrint-TouchUI #449: Icons Broken in OctoPrint 1.5.0 - Probably FontAwesome Conflict

jneilliii, 2020-11-21:
OctoPrint 1.5.0 is being shipped with FontAwesome 5.15.1 and I believe the way that you are referencing the content is no longer viable or mismatched.  This is what they look like.

[image]

cp2004, 2020-11-21:
Done some brief digging on this - seems that the font files are being reference with `FontAwesome`, which has now changed to `FontAwesome 5 Free` (or brands). Not sure of an easy way to adjust this that is not backwards-compatible. 
Touch UI also breaks icons it doesn't touch, for example the ones that are in the footer - since it applies its custom font family across them as well, which turns them to little boxes.

cp2004, 2020-11-21:
I suppose touchUI could bundle FA5 webfonts, for installs on OctoPrint <1.5.0. That might create strange issues, but then you would be able to change all the references like this to `Font Awesome 5 Free` (and obviously reference the font family file at the top of the file), without interfering when touchUI is not loaded.

BillyBlaze, 2020-11-22:
Thanks for reporting; this is fixed on the maintenance branch.

(commit message has the wrong version, it's 'Support for OctoPrint 1.5.0rc')

cp2004, 2020-11-22:
Looks great, just tried it. Was about to report #450 as well, but you've beaten me to it... 🙂

closed 2020-11-30 by BillyBlaze

devildant, 2020-12-01:
hello, il try maintenance branch and last release same problem :(

commit 27710714 (2020-11-22): :art: Support OctoPrint 1.4.0rc

OctoPrint upgraded their FontAwesome to 5.0, which breaks the icons set by TouchUI;
therefor reintroduce a local FontAwesome 4.7 file and rename FontAwesome to FontAwesomeForTouchUI
to prevent conflicts

This also replaced all hard-coded FA contents to use variables so that in the future its easier to
see what the name of the icon was/is

Fixes #449