BillyBlaze/OctoPrint-TouchUI #453: Browser Error

CodingPupper3033, 2020-11-26:
Hello!

First of all, I am not very experienced with Linux nor raspberrian, so you may have to help me a bit more than usual.

I am installing Touch-UI for the first time, went through all the downloads and drivers.

Upon the first "successful boot" (I may have skipped a few steps) It confronts me with some debug stuff (maybe the console, again no pi knowledge), a screen with touch ui in blue then after a while this image for about 5 seconds (below), then a red touch ui screen.

http://imgur.com/gallery/UiZf9k4

As an IT guy, and a helper for another niech program, I understand that I haven't provided enough information, and it is really frustrating. If you need something nit provided please ask.

I also was debating giving my log file, I would like to know if there is anything I should cut out, like maybe an api key shows up or something before I upload it.

Sorry to bother you, I spent 6hours trouble shooting and checking internet/previous closed issue s to no avail...

CodingPupper3033, 2020-11-26:
By skipping a few steps I am referring to the not successful boots. I double triple checked to make sure I didn't miss any steps before I put this issue in

CodingPupper3033, 2020-11-26:
I should have added, part of my troubleshooting was to disable auto logging in, this brought me to the normal login page, which after entering the correct login information will not do anything, no prompt, refresh, anything. It will tell me if I am entering wrong credentials however. I thought that was unusual behavior and may further towards a solution

JMHaro, 2020-12-04:
I'm getting the exact same thing also with a new setup that I'm trying to do for my first one adding a touch screen to my Octopi.  I'm interested in any help with this also.

TrevorWesterdahl, 2020-12-06:
I got that browser error when I added ::1 to the localNetworks section of the .octoprint/config.yaml file. Removed it and the error went away; however, I still can't get Touch UI to autologon locally, on the raspberry PI. It always brings up logon screen and there is no touch keyboard. Here is the relevant section:
accessControl:
    autologinAs: username
    autologinLocal: true
    localNetworks:
    - 127.0.0.0/8
    - 192.168.1.0/24
    salt: 8WlRaAs4UwGFnWDnWj1KcfBa2Bme4TcV

aesolutionsx3, 2020-12-08:
getting the same error.. fix?

imbrianj, 2020-12-08:
@aesolutionsx3 https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/445#issuecomment-732349663

aesolutionsx3, 2020-12-08:
fixed it. thx

JMHaro, 2020-12-09:
I just tried that fix #445.  Now mine goes to the login prompt.  I've tried rebooting it several times and even a full shutdown and restart but still goes to the login now.  I know I've been trying every other fix that I've been seeing so I don't know if something else I changed affected this, but any idea on what I can do to get past the login?

JMHaro, 2020-12-09:
I was able hook up a keyboard and log in and it is working now but still want's to return to the login each time even if I had the remember clicked on.

CodingPupper3033, 2020-12-14:
Hello again, I did not realize there are responses, thanks so much for them! I shall try these solutions, and hopefully, I have successfully turned on notifications this time. I appreciate all of your help!

CodingPupper3033, 2020-12-14:
Fix #445 Worked perfectly for me, thank you for helping, this is so amazing! You guys are the best, enjoy the holidays! I am going to leave this open as another fellow seems to be having some issues still.

closed 2021-01-05 by github-actions[bot]