BillyBlaze/OctoPrint-TouchUI #454: Username/Password Not Working

mikecoscia, 2020-11-26:
Not sure what I am doing wrong, but I am sitting at the please log in screen on my OctoPi and TouchUI will not accept my OctoPrint login credintials.  The screen just flashes and everything blanks out again.  I even tried changing the password to something simple.  Any ideas?

mikecoscia, 2020-11-26:
So...I tried installing TouchUI before upgrading OctoPrint from 1.3.12 to 1.4.2 and the TouchUI dashboard actually showed up and was usable for a minute or two!  Then it went right to the startup failed.  I tapped to retry and it never loaded up again.  Just stuck at connecting to TouchUI.

jasonstory, 2020-12-01:
Same happened to me and now octoprint 1.5 requires username and pw.  When I reinstall touch UI it asks for my username but now password.  Any suggestions @BillyBlaze ?

SirLonestar, 2020-12-07:
The same here.
 I have deactivated auto-login, now the octo-print login is displayed. If I enter the correct login data, the screen goes black for a moment and then I see the login again. If I enter incorrect login data, it show me that they are incorrect.

Maybe a problem with the session-management of octoprint?

closed 2020-12-29 by github-actions[bot]