BillyBlaze/OctoPrint-TouchUI #467: After fresh install screen ends up with Chrome connection error

JPElectron, 2020-12-22:
Fresh install on a Pi 4B, after upgrading things ending up at OctoPrint v1.5.2, OctoPi v0.18.0, and TouchUI v0.3.17

To get past the touchscreen (IO header connected, not HDMI connected) only displaying console text and not the browser I followed...

https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/251

and...

https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/365#issuecomment-595192476

...which you say DON'T change the URL param, and instead follow...

https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/368

and then doing this...

~/oprint/bin/octoprint config set --bool server.allowFraming true
sudo service octoprint restart
sudo service touchui restart

then I end up with a different error in chrome...

Loading TouchUI...
Startup failed, touch to retry
(if I touch, it goes back to Connecting to TouchUI and stays there forever)

Suggestions?

JMHaro, 2020-12-26:
I've gone through several installs myself and changing the TOUCHUI_URL as per #445 is what always resolves the issue for me.  Here are my notes for my own setup which has worked every time for me so far.  And I'm on the latest Pi version and Octoprint version currently.

		git clone https://github.com/BillyBlaze/OctoPrint-TouchUI-autostart.git ~/TouchUI-autostart/
		sudo ~/TouchUI-autostart/helpers/install
		Enter username in prompt:  JMHaro
		cd ~
		cd .octoprint/
		sudo nano config.yaml
		Underneath the autologinAs or autologinLocal lines add:
			    localNetworks:
		            - "127.0.0.1/8"
		            - "::1/128"
		nano ~/TouchUI-autostart/chromium.xinit
			In the CHROME_ARGS= line, Change the $TOUCHUI_URL   to   http://0.0.0.0/

		Disable the TouchUI Screen saver:
			sudo nano /etc/default/touchui
				Change the DISABLE_SCREENSAVER=false  to   DISABLE_SCREENSAVER=true
		sudo reboot now

lewisd32, 2021-01-03:
I have the same issue, and the fix described by @JMHaro does fix it for me.
Without changing the `$TOUCHUI_URL` in the chrome args, it didn't fix it. But with that, it did.

TheMrCaveman, 2021-01-03:
MM have the same issue but when issueing the nano ~/TouchUI-autostart/chromium.xinit command there is no CHROME_ARGS line?
what am i doing wrong here

mrouillard, 2021-01-06:
> MM have the same issue but when issueing the nano ~/TouchUI-autostart/chromium.xinit command there is no CHROME_ARGS line?
> what am i doing wrong here

double check, at the time of writing it is line 93.

closed 2021-01-29 by github-actions[bot]