BillyBlaze/OctoPrint-TouchUI #469: UI does not render correctly on low resolution.

t-keith, 2020-12-27:
[image]

Tried in chrome and Firefox. Version 0.3.17 480x854 screen resolution.

DanielYWoo, 2021-01-01:
The icons are not loaded, looks like network loading issue. Did you ever clean cache and restart the browser?

t-keith, 2021-01-03:
> The icons are not loaded, looks like network loading issue. Did you ever clean cache and restart the browser?

I tried with multiple browsers, including ones that have never loaded it.

DanielYWoo, 2021-01-04:
I experienced the same thing, but eventually the icons are loaded and it works again. To load the page, it takes 13MB to load, sometimes very slow on a Pi1 or Pi2B.

t-keith, 2021-01-04:
> I experienced the same thing, but eventually the icons are loaded and it works again. To load the page, it takes 13MB to load, sometimes very slow on a Pi1 or Pi2B.

I don't think that's what's going here. I did clear the cache, and icons are loaded, they just don't have enough room to display, or are too large to fit. If you look at the layout in the controls tab it is all messed up,nothing to do with icons.

DanielYWoo, 2021-01-05:
maybe you can open the developer tools in chrome to see which request are always pending or failed?

closed 2021-01-27 by github-actions[bot]