BillyBlaze/OctoPrint-TouchUI #470: OctoPrint Will Not Launch on Pi Display

AngeloAna, 2020-12-30:
I installed Touch UI according to the directions on a Pi 4 b with a MageDok 7 inch touchscreen. It gets to the Touch UI Loading OctoPrint page. At the bottom it is loading 127.0.0.1. It times out and says it can't load the page. Below is most of the SSH session showing pertinent information. I can launch it from the Pi desktop chromium browser with no problems.

pi@octopi:~ $ git clone https://github.com/BillyBlaze/OctoPrint-TouchUI-autostart.git ~/TouchUI-autostart/
Cloning into '/home/pi/TouchUI-autostart'...
remote: Enumerating objects: 330, done.
remote: Total 330 (delta 0), reused 0 (delta 0), pack-reused 330
Receiving objects: 100% (330/330), 58.94 KiB | 370.00 KiB/s, done.
Resolving deltas: 100% (193/193), done.
pi@octopi:~ $ sudo ~/TouchUI-autostart/helpers/install
[sudo] password for pi:
 ╔╦╗╔═╗╦ ╦╔═╗╦ ╦╦ ╦╦
  ║ ║ ║║ ║║  ╠═╣║ ║║
  ╩ ╚═╝╚═╝╚═╝╩ ╩╚═╝╩
Install helper v0.0.3

Boot to command line:
 - success

Running apt-get update (this will take a while)
Do NOT switch off the Pi or close this console until done!
 - success

Installing dependencies (xinit xinput xserver-xorg xserver-xorg-video-fbdev x11-xserver-utils matchbox unclutter chromium-browser):
 - success

Creating xserver config file:
 - success

Move touchui.init to '/etc/init.d/touchui':
 - success

Move touchui.default to '/etc/default/touchui':
 - success

Register touchui:
 - success

/home/pi/TouchUI-autostart/helpers/install-steps/6-setup-autologin-as: line 8: [: too many arguments
Skipping setup autoLogin for localhost:
 - Cancelled by user

TouchUI is successfully installed, Do not remove the directory
~/TouchUI-autostart/ and if you change the path of this directory
then edit the file /etc/default/touchui with the new path.

Launching TouchUI:
 - success
pi@octopi:~ $ sudo reboot

pi@octopi:~ $ sudo ~/TouchUI-autostart/helpers/install
[sudo] password for pi:
 ╔╦╗╔═╗╦ ╦╔═╗╦ ╦╦ ╦╦
  ║ ║ ║║ ║║  ╠═╣║ ║║
  ╩ ╚═╝╚═╝╚═╝╩ ╩╚═╝╩
Install helper v0.0.3

Skipping installation of dependencies:
 - All dependencies are already installed

Skipping creation of xserver config file:
 - /dev/fb0 already configured

Move touchui.init to '/etc/init.d/touchui':
 - success

Move touchui.default to '/etc/default/touchui':
 - success

Register touchui:
 - success
[... 498 lines cut ...]
[    21.542] (**) SIGMACHIP USB Keyboard Consumer Control: always reports core events
[    21.542] (**) Option "Device" "/dev/input/event1"
[    21.542] (**) Option "_source" "_driver/libinput"
[    21.543] (II) libinput: SIGMACHIP USB Keyboard Consumer Control: is a virtual subdevice
[    21.543] (**) Option "config_info" "udev:/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3:1.1/0003:1C4F:0002.0002/input/input1/event1"
[    21.543] (II) XINPUT: Adding extended input device "SIGMACHIP USB Keyboard Consumer Control" (type: KEYBOARD, id 10)
[    21.543] (**) Option "xkb_model" "pc105"
[    21.543] (**) Option "xkb_layout" "gb"

pi@octopi:~ $ xinit /home/pi/OctoPrint-TouchUI-autostart/start_chromium.xinit

(EE)
Fatal server error:
(EE) Server is already active for display 0
        If this server is no longer running, remove /tmp/.X0-lock
        and start again.
(EE)
(EE)
Please consult the The X.Org Foundation support
         at http://wiki.x.org
 for help.
(EE)
XIO:  fatal IO error 11 (Resource temporarily unavailable) on X server ":0"
      after 7 requests (7 known processed) with 0 events remaining.
pi@octopi:~ $

modusmoron, 2020-12-30:
Try this fix: [https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/445](https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/445)

I've been battling this for the past two days, just found that solution, it worked great!

AngeloAna, 2020-12-30:
That fixed the problem. Thank you very much!

closed 2020-12-30 by AngeloAna