BillyBlaze/OctoPrint-TouchUI #474: Cannot connect to TouchUI after Octoprint update

bdiesel1369, 2021-01-03:
I get a white screen error of localhost refused to connect after upgrading OctoPrint to recent release. I have added allowFraming: true and localNetworks: fixes to respective files and cannot get the localhost refused to connect to work.

saronatart, 2021-01-12:
Hi, I have the same problem.
No one a solution?

yet-another-average-joe, 2021-01-13:
sudo nano ~/TouchUI-autostart/chromium.xinit

In CHROME_ARGS, replace touchui localhost IP with http://0.0.0.0
You may also want to disable pinch if you have a capacitive multitouch : add --disable-pinch

This what my autostart looks like 

`
CHROME_ARGS="	--no-first-run --kiosk $TOUCHUI_ARGS $TOUCHUI_DEBUG
				--dns-prefetch-disable --disable-sync-preferences --disk-cache-size=1 --disk-cache-dir=/dev/null
				--disable-java --disable-plugins --disable-extensions --disable-infobars
				--user-agent='TouchUI (X11, Chrome $CHROME_VERSION$OCTOPI) (P:$PORT)'
				--start-maximized --window-position=0,0 --user-data-dir=$CONFIG_DIR
				--disable-pinch
				http://0.0.0.0"
`

bdiesel1369, 2021-01-31:
Added the suggested lines and got a different error message. See attached picture.
[image]

yet-another-average-joe, 2021-02-01:
I realize the command line is not correct : 

`sudo nano ~/pi/TouchUI-autostart/chromium.xinit`

makes no sense ! It's obviously :

`sudo nano ~/TouchUI-autostart/chromium.xinit`

Normally http://127.0.0.1 and http:://0.0.0.0 work (just tested again). And why isn't OctoPrint service running ????

Did you try :

```
sudo service octoprint start or restart
sudo service touchui start or restart
```

And what the log file tells to you ?
 
A wrong URL in TouchUI does not prevent octoprint service from starting (AFAIK). There might be something else...
I will not be able to test again, as I'm reflashing with the new OctoPi 0.18.0, and will not install TouchUi autostart (not needed with larger screens (7" here : TouchUI can be toggled in OctoPrint menu - witch is impossible with the autostart)

Maybe you should consider that. Your screen looks like it's a 7" or so. Just boot with Chromium in kiosk mode.

closed 2021-02-09 by github-actions[bot]

bdiesel1369, 2021-02-14:
My screen is a 4 inch THT I/O screen I am not using HMDI to connect to screen. After making changes I rebooted raspberrypi, which should have restarted touchui service I will have to include the log file next.