BillyBlaze/OctoPrint-TouchUI #476: Toggled to Touch UI - can't toggle back

wendli01, 2021-01-18:
This is a duplicate of #312.

Apparently, Touch UI is too broken to deactivate it.
My solution was to delete the [custom css file](https://github.com/BillyBlaze/OctoPrint-TouchUI/blob/master/octoprint_touchui/static/css/touchui.css) via the developer console of my browser.
This temporarily resets the modifications and makes octoprint usable again, allowing for deactivation of Touch UI.

An alternative solution is the one proposed in #468, however it requires ssh access to the octoprint server and a restart, interrupting ongoing prints.

A long-term solution is either fixing Touch UI or removing it from the plugin repository .

yet-another-average-joe, 2021-01-23:
Or just don't install the autorun !

This is what I did ; can't remember where I found this, as I write down the good tricks, and often forget to write the links... ; did it because I don't want them for development purpose. Probably OctoPrint offcial forum, a tutorial by Foosel ; needed some trial and error and will redo from start for confirmation (maybe will make a tutorial) :

Fresh OctoPi install
Install X with no bloat
Install TouchUI with no autorun

and then you will be able to choose to log to command line (OctoPrint service being running anyway), or to OctoPrint, or to TouchUI. Or even to any other OctoPrint frontend (OctoDash, etc.)

Install xorg server and the OpenBox window manager with NO bloat :
`sudo apt-get install --no-install-recommends xserver-xorg x11-xserver-utils xinit openbox`

Install Chromium with no bloat :
`sudo apt-get install --no-install-recommends chromium-browser`

Now edit the OpenBox autostart :
`sudo nano /etc/xdg/openbox/autostart`

add :

`# Disable any form of screen saver / screen blanking / power management
xset s off
xset s noblank
xset -dpms

# Allow quitting the X server with CTRL-ATL-Backspace
setxkbmap -option terminate:ctrl_alt_bksp

# Start Chromium in kiosk mode
sed -i 's/"exited_cleanly":false/"exited_cleanly":true/' ~/.config/chromium/'Local State'
sed -i 's/"exited_cleanly":false/"exited_cleanly":true/; s/"exit_type":"[^"]\+"/"exit_type":"Normal"/' ~/.config/chromium/Default/Preferences
chromium-browser --disable-infobars --kiosk 'http://0.0.0.0'
`

You could also add --disable-pinch and other options to Chromium.

Now you can :

- autostart your OctoPrint server with no X when powering on = command line
- launch OctoPrint with 'startx'
- chose the option from within OctoPrint AND/OR TouchUI : normal/touchui (the option is obviously persistent)
- decide to have an autostart to graphics environment from raspi-config : chose command line or desktop

You can do exactly the same you do on a PC.

End of story.

Removing TouchUI from the plugin repository ? Are you kidding ? 5mn with Google are enough for the solution to this problem.

IMHO the TouchUI autostart was made for lazy users, and for Linux beginers. I belong to these two categories. But it has its HUGE drawbacks. A couple years ago, it had to be installed by hand ; it was much more dicdactic. Linux allows many things, just do them (Win32 API fanboy inside !)

wendli01, 2021-01-23:
@yet-another-average-joe The issue is not with Touch UI's autorun but with its interface.

As stated before, it breaks the UI in a way that does not allow for disabling it from the octoprint web interface.

My fix, as others, is a hotfix for _deactivating_ Touch UI - it is not a problem with user installation or configuration.
This is reflected in the title of this issue - "can't __toggle__ back" - also a duplicate of #312 and many other issues.

yet-another-average-joe, 2021-01-23:
Your fix is not a fix, because there's nothing to fix :  you installed a lo-res kiosk OctoPrint wrapper, and you'd like to use it as a standard browser. 
You can't toggle back because you are using TouchUI as a standalone specialized application, intended for very low res TFTs, and the browser makes no sense on such devices. Would you ask for other front ends like OctoDash, OctoScreen or OctoPiTouchPanel to be removed from the database because of the exact same thing ?
You missed the point : if you have to switch between a browser and TouchUI, install X, some window manager, whatever browser  you like and TouchUI plugin : you don't even have to install Chromium. Works the same with Firefox ! The autorun has always been optionnal !
It's easy, straightforward and will give you full control.

wendli01, 2021-01-23:
Please refer to the documentation:

`This plugin will transform the OctoPrint layout into a Mobile/TFT friendly layout.`

`The interface will automatically start when your browser is smaller then 980 pixels in width or if you’re browsing with a touch device.`

```
Supported browsers
Chrome 30+
Firefox 40+
Safari Mobile
Chrome Mobile
```

I will amend my initial proposal to _fixing Touch UI or adapting the plugin description to reflect that it is __only__ suitable for low-res TFTs._

yet-another-average-joe, 2021-01-23:
It's true the documentation is a bit unclear !
Even on high resolution mobile devices (phones), OctoPrint is barely usable without TouchUI. Here's a crappy video I just shot to show you : 7" 1024x600 capacitive touch. Don't take into account the "false touches", it's me, not the device (capacitive devices don't like my fingers)
OctoPrint is installed as a plugin, with no autorun. The browser is launched from SSH or a keyboard, or with the startup option in raspi-config. The standard OctoPrint UI is difficult to use (tiny buttons and menus). No problems to go back and forth between the two UIs (OctoPrint and TouchUI)
(the Marlin Mode is a work in progress)

https://youtu.be/j3KQbSgojR0

yet-another-average-joe, 2021-01-24:
I retrieved the tutorial I was using when starting with the "no autorun TouchUI" setup : https://die-antwort.eu/techblog/2017-12-setup-raspberry-pi-for-kiosk-mode/

After a lot of verifications, and some adjustments, you can do this, and you will get the best of the two worlds (command line, OctoPrint and TouchUI - yes it's three) :

From a fresh install, we need the X server (xorg), a window manager (will be openbox), a login manager (will be lightdm), a browser (will be Chromium) and something that will help us to get rid of the stupid mouse cursor :

```
sudo apt-get update
sudo apt-get upgrade
sudo apt-get dist-upgrade
sudo apt-get lightdm
sudo apt-get unclutter
sudo apt autoremove
sudo apt-get install --no-install-recommends xserver-xorg x11-xserver-utils xinit openbox
sudo apt-get install --no-install-recommends chromium-browser

```
Now we have to tell X and chromium how to start : we have to edit the openbox autostart ; it could be ~/.xinit, and I'd prefer, but I get black margins (looking like what does overscan). Too lazy to investigate... So :

`sudo nano /etc/xdg/openbox/autostart`

add these lines :

```
# Disable any form of screen saver / screen blanking / power management
xset s off
xset s noblank
xset -dpms
# Allow quitting the X server to CLI with CTRL-ATL-Backspace
setxkbmap -option terminate:ctrl_alt_bksp
# Things I don't really understand...
sed -i 's/"exited_cleanly":false/"exited_cleanly":true/' ~/.config/chromium/'Local State'
sed -i 's/"exited_cleanly":false/"exited_cleanly":true/; s/"exit_type":"[^"]\+"/"exit_type":"Normal"/' ~/.config/chromium/Default/Preferences
# hide mouse cursor
unclutter -idle 0.01 -root
# open OctoPrint webpage : no pinch, no bars, no scrollbars
# http://0.0.0.0 because localhost = log loop (known issue, easy fix)
chromium-browser --enable-features=OverlayScrollbar,OverlayScrollbarFlashAfterAnyScrollUpdate,OverlayScrollbarFlashWhenMouseEnter --disable-infobars --disable-pinch --disable-overlay-scrollbars --kiosk 'http://0.0.0.0'
```

Then, you can choose how you want to boot ; raspi-config :

- console (autologin or not) ; then you can start the browser typing "startx"
- browser : select desktop/autologin, or you will have to type ID+PSWD... Not convenient on a touch device with no real or virtual keyboard, but VNC will do it !

Something else I find usefull (it is for apps writing directly to the frame buffer) ; the X-Terminal ; if you code bare metal, it is a must have :

`sudo apt-get install xterm`

And usefull parameters for VNC :

`sudo nano /root/.vnc/config.d/vncserver-x11`

add :

```
CaptureMethod=1
CaptuteTech=raspi
IdleTimeout=0
```

VNC will will never close the connection.

closed 2021-02-15 by github-actions[bot]

wendli01, 2021-02-15:
I would like to keep this issue open as it is not resolved.

This can serve as a hotfix in the interim.