BillyBlaze/OctoPrint-TouchUI #480: TouchUI Stuck at Loading Screen

SubieMom, 2021-02-02:
TouchUI Errors with "Browser Error Console" Touch to try again. Then upon touching it reloads and never gets past the loading screen. There are trouble shooting methods online but I cannot SSH into my Octopi as I never set SSH up. The unit is set to AutoStart TouchUI upon boot as the directions stated to do that as I could change it later. 

Is there a way for me to get back to the terminal/command line so I can troubleshoot or am I going to have to reflash octopi?

CodeMonkeyX, 2021-02-02:
I thought it was just me! Having the exact same behavior. I am not sure when this started, because 90% of the time I use the web interface, but the other day I was at the machine and could not use TouchUI.

For me OctoPrint is still running and working fine in the background, I can still connect via a web browser from a different computer on the network.

I tried uninstalling the plugin and reinstalling.

@SubieMom you should be able to enable ssh by removing the SD card, and add a file called "ssh" in the root folder. Then you should be able to SSH in when you reboot the PI. https://www.raspberrypi.org/documentation/remote-access/ssh/

But yeah any help debugging this, what logs to look at etc, would be appreciated.

SubieMom, 2021-02-02:
So I ended up just trying to SSH in anyway and even though I hadn't explicitly set it up it let me in. I completely (well I messed up and didn't uninstall like 5 things) uninstalled TouchUI. Judging by the lack of response on the other issues in here, I can't wait for this to get resolved and will have to go with a different interface. 😟

fluffymonster, 2021-02-05:
Feel your pain @SubieMom. My big concern is a update to OctoPrint will break it and will not be fixed. 
Had a look at [OctoScreen](https://github.com/Z-Bolt/OctoScreen). not as suitable for my use as TouchUI but a support channel is critical. I need/want portrait mode.

anibluny, 2021-02-08:
Had the exact same issue with the only warning in octoprint.log showing "tornado.access - WARNING - 403 POST /api/settings"
Resolved by:
action:
sudo nano etc/default/touchui

then edit the line:
TOUCHUI_PORT="8888"

and change it to:
TOUCHUI_PORT="80"

[https://community.octoprint.org/t/found-solution-combined-for-octopi-0-18-0-with-touchui-and-not-connecting-to-1/28765](Credit to BjornTHV for the temp workaround). Hope this works for you.

CodeMonkeyX, 2021-02-08:
Thanks. I thought it would be something relatively simple, because
Octoprint still worked fine, the browser still loaded etc.

I can't test it because I decided to give another front end a shot anyway,
but it's good to know I can fix the issue if I got back.

iguana007, 2021-02-10:
> Had the exact same issue with the only warning in octoprint.log showing "tornado.access - WARNING - 403 POST /api/settings"
> Resolved by:
> action:
> sudo nano etc/default/touchui
>
> then edit the line: [...]

Thank you, you are my hero :)

I am running OctoPi on: Raspberry Pi 3 Model B Plus Rev 1.3 with Waveshare 4" HDMI LCD display, IPS, 800x480.

I just did fresh re-install of completele solution: Octopi and TouchUI, following installation instructions, but I got stuck on the same error.

In my case, only difference was, that config file was located at: /etc/default/touchui

There I changed port to 80 and after reboot is everything fine :)

Many thanks!

patoberli, 2021-02-10:
In my case manually editing the chromium.xinit file as per here: https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/445 fixed the issue for me. 
Please fix that in the source.

closed 2021-03-04 by github-actions[bot]