BillyBlaze/OctoPrint-TouchUI #484: TouchUI (OctoPrint on Windows 10) Folder path adding double backslash to file path

RadicalxEdward, 2021-02-15:
I've been having issues with OctoPrint disconnecting mid-print and then not being able to resume and the print being ruined. I tried to enable Serial.log since that seems to be the first thing people ask to see, but when doing so i get an error from TouchUI that says:

```
TouchUI: Whoops, something went wrong...
[Errno 13] Permission denied: 'c:\\program files\\octoprint\\venv\\lib\\site-packages\\octoprint_touchui/static/css/touchui.custom.5000.css'
```

That path doesn't exist on windows. The correct path is C:\Program Files\OctoPrint\venv\Lib\site-packages\octoprint_touchui\static\css\touchui.custom.5000.css

For some reason TouchUI is adding double backslashes to the windows file path and then using forward slashes within the plugin folder. Don't know why, don't know what to do about it.

BillyBlaze, 2021-02-15:
Thanks for reporting; this should be fixed using `os.path` within Python.

The double backslashes is what Windows is expecting. And the forward slashes in hard-coded by TouchUI. So on Windows the forward slashes should be converted to double backslashes.

RadicalxEdward, 2021-02-15:
> Thanks for reporting; this should be fixed using `os.path` within Python.
> 
> The double backslashes is what Windows is expecting. And the forward slashes in hard-coded by TouchUI. So on Windows the forward slashes should be converted to double backslashes.

Thank you for the information. I'm not super familiar with Python so i'll have to research that. 

I also noticed that installing plugins is giving me a similar error about permissions. I guessed that it had to do with me installing to C:/Program Files. I gave the "Users" group Full control access to the OctoPrint root folder and I'm no longer getting those errors. I'll have to keep an eye out about TouchUI specifically. The issue may not have been anything to do with TouchUI specifically and it was just the first to tell me about it. 

Im not knowledgable enough to speak to the security of my solution (i'm gonna go ahead and guess it's in the ballpark of "not secure at all" but i'll be moving over to a Raspberry Pi soon anyway.

Should I go ahead and close this thread?