BillyBlaze/OctoPrint-TouchUI #486: Touchui startup failing

olivereckel, 2021-02-20:
I am using a Raspberry 4 and a 7 inch original raspberry touchscreen. I installed following the "Setting up a Touchscreen" and "Boot to browser" instructions . It installed fine however will not start into TouchUI.

[image]

This is the error message, when I click on browsers error console it opens a help page in the browser, but I can't find an error log. I can navigate in the help page however.

Attached is my octoprint log

[octoprint.log](https://github.com/BillyBlaze/OctoPrint-TouchUI/files/6014813/octoprint.log)

Thanks for the help !!

Zanop, 2021-02-23:
Hi,

I have the same, looks like server.py gets stuck after the first couple of requests and responds very slowly.

Zanop, 2021-02-23:
It might be related to these

https://bugs.python.org/issue31639

https://bugs.chromium.org/p/chromium/issues/detail?id=195550

Cheers

olivereckel, 2021-02-24:
so how do I fix it? - or should I just move to Octodash

BenWhite-git, 2021-02-24:
HI, Also have this problem on Pi 4b.

maditso, 2021-02-24:
Same problem here, with RasPI3B+

Zanop, 2021-02-25:
> 
> 
> so how do I fix it? - or should I just move to Octodash

OMG, I just installed OctoDash, that's amazing, Thanks!

ab-tools, 2021-03-13:
I've exactly the same problem with a PI 4b, too.

Is there any fix for this or is OctoPrint-TouchUI simply no longer in active development and we should rather switch to OctoDash now?

Zanop, 2021-03-13:
Apparently it's abandoned. I already switched to OctoDash which is very nice, exactly what I wanted, and not a web page in chrome, which is overkill in terms of memory and cpu usage.

ab-tools, 2021-03-13:
Yeah, indeed, I also switched to OctoDash now:
Initially I was a bit worried that is would have less functionality, but it's actually great as you can still use the old/normal OctoPrint user interface remotely in a browser if you need more functionality and at the same time OctoDash on the PI touch screen.

This was one thing I did not like that much about TouchUI that you then have this same touch UI also when you open OctoPrint remotely from your Desktop computer in a web browser (yes, I know, you could toggle between the UIs, .

Therefore, yes, OctoDash seems indeed to be the best solution! :-)

closed 2021-03-21 by github-actions[bot]