BillyBlaze/OctoPrint-TouchUI #487: TouchUI Loads then Crashes

tibmeister, 2021-02-26:
TouchUI loads, the interface pops up with some boxes for the icons, then goes to white screen and starts loading TouchUI all over again.  No errors or anything in the octoprint.log, so not sure what's going on.
Fresh install (Octoprint 0.18.0) on rPi3B+

tibmeister, 2021-02-26:
Turns out it's the UI Customizer plugin that breaks TouchUI.  Very disappointed, I've used TouchUI for some time now, > 2 years, but it seems that I cannot use it anymore unless this issue get's fixed.

brynnflynn, 2021-03-05:
I'm not so sure it's just that plugin--I'm getting similar behavior. No errors logged, and TouchUI seems to get through the connecting to TouchUI, loading octoprint, and loading TouchUI steps, but then crashes to a white screen which eventually returns to the TouchUI "error, tap to retry" screen. Attached are my logs from my last boot.
[octoprint.log](https://github.com/BillyBlaze/OctoPrint-TouchUI/files/6088088/octoprint.log)

MLutzAT, 2021-03-10:
same problem here, with uicustomizer on, touchui constantly crashes with a white screen 1-2s after loading...
when turning uicustomizer off, touchui works as expected!

is there any known fix to that?

MLutzAT, 2021-03-12:
another problem, sorry for posting it here, but the relevant thread/issue has been locked: i'm also monitoring that chromium puts heavy stress on my pi, i think i'm almost unable now to use it without active cooling. can it be that this is all linked together, and maybe an other version of chromium would make more sense in terms of stability and/or CPU usage?
please see at issue #174 for details about the CPU-usage, it startet a few years back and never got sorted...

larp-welt, 2021-03-24:
> same problem here, with uicustomizer on, touchui constantly crashes with a white screen 1-2s after loading...
> when turning uicustomizer off, touchui works as expected!

Same for me! :-(

larp-welt, 2021-04-04:
It looks like there is a solution for this problem:

https://github.com/LazeMSS/OctoPrint-UICustomizer/issues/60

closed 2021-04-26 by github-actions[bot]