BillyBlaze/OctoPrint-TouchUI #488: Boot-to-browser with 3.5" LCD is stuck at "Connecting to TouchUI"

orrmany, 2021-03-13:
1. I've installed my 3.5" LCD, which works. 
2. I've installed TouchUI via the Octoprint plugin manager. I've followed https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Boot-to-Browser. The install helper reports all success. 
3. I can activate and de-activate and use TouchUI via the Web interface remotely. However, I can't access TouchUI locally, the LCD shows the "TouchUI" wire line logo in blue with the undersign: Connecting to TouchUI --- and it stucks here.... It shows this display forever

orrmany, 2021-03-13:
I've also checked Issue #84, it does not offer any relevant clue (since it just recommends to install via the helper script, which I did)

roachmotel3, 2021-03-15:
Same here -- piling on.  Have disabled ipv6, I'm able to auto-login to port 5000 -- but port 8888 just spins.

roachmotel3, 2021-03-15:
Also, it seems like there's a defect in the step6 script for the installer.  I had to modify it to eliminate an error, though it seems minor (the grep returns multiple results which leads to an error in the -z test further down on line 8:

`OCTOPRINT_EXEC=$(ps -ef | grep octoprint | grep -v grep | awk '{print $9}' | head -n 1)`

Also, the git dir is named "OctoPrint-TouchUI-autostart" but various scripts expect it to be "TouchUI-autostart".  A quick symlink fixed that, but still is broken.

Drzet, 2021-03-17:
https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/445#issuecomment-732349663

orrmany, 2021-03-17:
Thanks @Drzet . Meanwhile I've dropped the entire TouchUI and switched to OctoDash, but good to know the likely solution.

supawiz6991, 2021-04-01:
> [#445 (comment)](https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/445#issuecomment-732349663)
> 
> Fix here

HI
Just wanted to add that I too ran into this issue and like the others can confirm that the solution noted in [#445](https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/445#issuecomment-732349663), changing $TOUCHUI_URL to http://0.0.0.0 resolved the issue for me as well. 

Any idea if this is a bug or if its perhaps a misconfiguration some where?
Is it a b

closed 2021-04-09 by github-actions[bot]