BillyBlaze/OctoPrint-TouchUI #491: Touchui doesnt show on screen Startup Failed. Tap to retry

Meganoodle71, 2021-04-13:
Says it all above, thought I would treat myself to an official pi 7" touchscreen to fit to the printer. Ive spent two solid days trying to get this to work following various suggested fixes I found on google (re burning the Octopi image and starting from scratch each time to be sure).
But regardless to what i do I get Startup Failed,  or Browser's error in console, or touchui refused to connect,  there doesn't seem to be any logic behind which error pops up just random i guess.

I know this issue is quite an old one , Ive read a lot of forums. 
Other than rolling back my version of octopi Im not sure what else I can do.

Any guidance would be appreciated.

cheers

newnec1, 2021-04-14:
i get the same thing, i have 3 pis, all with 7in touchscreen and ended up erasing all 3 with fresh installs of octopi then install all OS updates and just installed the touchui plugin with the boot to browser and i get the samething as you on all 3.

Meganoodle71, 2021-04-14:
@newnec1  at least it proves its not something obvious then. I was starting to think it was me. Like you I have since tried the same setup on another Pi 3B and 2 different Pi 4's. Ive actually given up for the time being , its evident Im not going to get anywhere without assistance. So Ill wait and see what happens here.
Hope above hope I can get it resolved , the touchui looks great, failing that ill have to re purpose the screen I just purchased .

bonshates, 2021-04-14:
Having the same issue with a new install. Here's what I see at the end of the octoprint.log.
2021-04-14 21:36:14,228 - tornado.access - WARNING - 403 GET /api/settings (::1) 370.96ms

On the screen I get a red triangle with an exclamation mark. Below that is says "browser's error console       octoprint.log"

Any help would be greatly appreciated. 

Kudos to BillyBlaze for all his hard work on this.

bonshates, 2021-04-14:
> Having the same issue with a new install. Here's what I see at the end of the octoprint.log.
> 2021-04-14 21:36:14,228 - tornado.access - WARNING - 403 GET /api/settings (::1) 370.96ms
>
> On the screen [...]

I just tried the fix from Craigeous in https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/445#issuecomment-732349663 and it works now.

bonshates, 2021-04-14:
Hmmm. After applying the #445 fix my webcam no longer worked. I have restored a to known working version of octoprint on my Pi3 and will try TouchUI again in a few months. Hopefully this will have been fixed by then. Open for any suggestions in the mean time.

newnec1, 2021-04-15:
i tried the #445 and it didnt fix mine. i ended up just trying out octodash instead. im sure ill eventually go back to touchui but i cant waste anymore time right now on it

supawiz6991, 2021-04-15:
> I just tried the fix from Craigeous in https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/445#issuecomment-732349663 and it works now.

Just had this issue again after reinstalling octopi. #445 worked for me. Noted that in an different issue that auto closed just before you opened this one. No response from the dev. 
Who knows if this will get a proper fix.

closed 2021-05-08 by github-actions[bot]

BillyBlaze, 2021-07-04:
Thanks for reporting. The issue has been fixed.

If you made changes to your default file then restore `8888`
If you changed the script.js file then restore the original files by running `git reset --hard` in the TouchUI-autostart folder.

After that you can [update your bootloader](https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Update-your-bootloader) as described in the wiki.