BillyBlaze/OctoPrint-TouchUI #493: "Expand filesystem Check" does not work with Raspberry and USB Disk

TK90147091, 2021-04-22:
I run a Raspberry with usb disk and the "Expand filesystem Check" failed, but I have a lot of space free ...

Expand your file system before running this helper.
Run `sudo raspi-config` and select `Expand filesystem`

root@blackhole:/home/pi/TouchUI-autostart/helpers# df -h .
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       147G  3.3G  138G   3% /

Will-wastelander, 2021-04-22:
What exactly are you expecting to see here ? In addition, this would he a
raspberry OS issue, not a TouchUI issue..

FYI, looks like everything is fine with your Linux OS..

TK90147091, 2021-04-27:
Well, in my case, i deleted the "Expand filesystem Check" in TouchUI-autostart/helpers/install-steps/1-check-mounted-sd and run the installation and all was fine, but maybe there are other people in this world which get the same error. And how did you say, it is not an error of TouchUI, so remove the "Expand filesystem Check" from the source code and add an "It is enough space available" Test to the code ...

closed 2021-05-19 by github-actions[bot]