BillyBlaze/OctoPrint-TouchUI #494: OctoPrint 1.6 : messed up dosplay on TouchUI

yet-another-average-joe, 2021-05-01:
Hi,
weird issue... Yesterday I updated Octoprint, and the next day, this is what I get : microscopic icons and no menu bar. Rebooting has no effect. I cannot test further (for now) because the printer will be in use 24/7 for the 5 next days...

Am I alone with this issue ?

[image]

akrilov, 2021-05-01:
Yep, same issue here, right after updating to 1.6.0.
I have just a blank menu bar without any icons, and even in regular Octoprint UI the touchUI configuration button shrunk to like 30% of what it was before...

Tried multiple browsers, Firefox, Chrome, edge, Kindle's Silk, all produce the same result.

[image]

akrilov, 2021-05-01:
Here's the js console log from the browser, if that'll help:

packed_core.js?40786308:1223 Starting dependency resolution...
packed_core.js?40786308:1236 ... dependency resolution done
packed_core.js?40786308:1259 Initial application setup done, connecting to server...
packed_core.js?40786308:1062 Connected to the server
packed_core.js?40786308:1067 Triggering connect on all view models
packed_core.js?40786308:1069 Server (re)connect processed
packed_core.js?40786308:1259 Finalizing application startup
packed_core.js?40786308:1240 Going to bind 51 view models...
packed_plugins.js?3efa98e3:22841 octolapse.help.js - Binding help links to #octolapse_tab a.octolapse_help[data-help-url]
packed_core.js?40786308:1252 Did not bind view model BackupViewModel to target #wizard_plugin_backup since it does not exist
packed_core.js?40786308:1252 Did not bind view model UsageViewModel to target #wizard_plugin_tracking since it does not exist
packed_core.js?40786308:1252 Did not bind view model ThespaghettidetectiveWizardViewModel to target #wizard_plugin_thespaghettidetective since it does not exist
packed_core.js?40786308:1252 Did not bind view model SoftwareUpdateViewModel to target #softwareupdate_confirmation_dialog since it does not exist
packed_core.js?40786308:1252 Did not bind view model SoftwareUpdateViewModel to target #wizard_plugin_softwareupdate since it does not exist
packed_core.js?40786308:1252 Did not bind view model BLTouchViewModel to target settings_plugin_BLTouch_form since it does not exist
packed_core.js?40786308:1252 Did not bind view model bedlevelvisualizerViewModel to target #wizard_plugin_bedlevelvisualizer since it does not exist
packed_core.js?40786308:220 User Alex logged in
packed_core.js?40786308:225 Login mechanism: http
packed_core.js?40786308:1257 ... binding done
packed_core.js?40786308:1257 Application startup complete
6The resource <URL> was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.
/?#touch:1 The resource http://192.168.27.5/static/vendor/font-awesome-5.15.1/webfonts/fa-solid-900.woff?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.
/?#touch:1 The resource http://192.168.27.5/static/vendor/font-awesome-5.15.1/webfonts/fa-regular-400.woff?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.
/?#touch:1 The resource http://192.168.27.5/static/vendor/font-awesome-5.15.1/webfonts/fa-brands-400.woff2?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.
/?#touch:1 The resource http://192.168.27.5/static/vendor/font-awesome-5.15.1/webfonts/fa-brands-400.woff?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.
/?#touch:1 The resource http://192.168.27.5/static/vendor/font-awesome-5.15.1/webfonts/fa-solid-900.woff2?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.
/?#touch:1 The resource http://192.168.27.5/static/vendor/font-awesome-5.15.1/webfonts/fa-regular-400.woff2?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.
2packed_plugins.js?3efa98e3:22841 octolapse.help.js - Binding help links to .octolapse .webcam_settings a.octolapse_help[data-help-url]
packed_core.js?40786308:449 Terminal: Detected very slow client (needed 415ms for processing new log data), completely disabling terminal output during printing
TerminalViewModel.self._disableUnfancy @ packed_core.js?40786308:449
TerminalViewModel.self.fromCurrentData @ packed_core.js?40786308:451
callViewModelIf @ packed_core.js?40786308:1192
(anonymous) @ packed_core.js?40786308:1186
Pn @ packed_libs.js?86afb3b9:30
(anonymous) @ packed_libs.js?86afb3b9:51
callViewModelsIf @ packed_core.js?40786308:1186
callViewModels @ packed_core.js?40786308:1185
(anonymous) @ packed_core.js?40786308:1069
DataUpdater.self._ifInitialized @ packed_core.js?40786308:1083
DataUpdater.self._onCurrentData @ packed_core.js?40786308:1069
(anonymous) @ packed_client.js?0a0190d8:20
Pn @ packed_libs.js?86afb3b9:30
(anonymous) @ packed_libs.js?86afb3b9:51
OctoPrintSocketClient.propagateMessage @ packed_client.js?0a0190d8:20
(anonymous) @ packed_client.js?0a0190d8:27
(anonymous) @ packed_libs.js?86afb3b9:47
_t @ packed_libs.js?86afb3b9:35
(anonymous) @ packed_libs.js?86afb3b9:46
(anonymous) @ packed_libs.js?86afb3b9:51
onMessage @ packed_client.js?0a0190d8:27
r.dispatchEvent @ packed_libs.js?86afb3b9:1599
(anonymous) @ packed_libs.js?86afb3b9:1599
w._transportMessage @ packed_libs.js?86afb3b9:1599
i.emit @ packed_libs.js?86afb3b9:1599
ws.onmessage @ packed_libs.js?86afb3b9:1599

yet-another-average-joe, 2021-05-01:
Reverting to OctoPrint 1.5.3 solves the problem :

`~/oprint/bin/pip install https://github.com/OctoPrint/OctoPrint/archive/1.5.3.zip`

On a remote computer, if TouchUI was activated, erasing the browser cache solves the problem.

avigeilpro, 2021-05-02:
same issue here, but only with customization enabled.

blackbird042, 2021-05-03:
As stated above .. for me too, this only happens when customization is enabled

mosestpi, 2021-05-05:
Disabling / Tweaking "Floating Navbar" plugin solved my problem

blackbird042, 2021-05-05:
I've tied this on a fresh/empty OctoPrint install with TouchUI as the only extra plugin installed.
As soon as I enable customization in TouchUI the problem arises.

Nailig, 2021-05-14:
I just re-installed octopi and touchui and using a 800 x 480 screen and i have no display issues.
Perhaps some chache or one-time generated file that is the issue?

kromix84, 2021-05-14:
> I just re-installed octopi and touchui and using a 800 x 480 screen and i have no display issues.
> Perhaps some chache or one-time generated file that is the issue?

Weird, 
just toggled off everything in touch UI including the Enable customizations in the pluggin settings, then reloaded the service, toggled everything back on and the nav bar works. but it wont let me customize the Nav color anymore.  the terminal change works

could be that the nav color is pointing to deprecated variable that was broken in the 1.6 patch? 

in any case my Pi looks now default instead of the red nav bar.

Jon-So, 2021-05-18:
I just did the same thing as Kromix84, I uncheched the "enable Customization" option and the buttons came back....

crzy1, 2021-06-03:
I just tried everything listed above and there was no change with the issue.  However when I turned off customization touch UI turned off on my browser but only the print options and file listings remained.  Still cant get any menu options.

BillyBlaze, 2021-06-03:
thanks for reporting, I will have a look at this as soon as possible.

Thierry-Myron, 2021-06-18:
Same issue
I disabled the Customisation for TouchUI to work again!

avigeilpro, 2021-06-25:
hi, any update on this problem ?

BillyBlaze, 2021-07-04:
The issue has been fixed on the maintenance branch and will be released a.s.a.p

BillyBlaze, 2021-07-05:
0.3.18 has just been released with this fix

closed 2021-07-06 by BillyBlaze

yet-another-average-joe, 2021-07-06:
Installed yesterday, with the OctoPrint update, as soon as I receided the notification.
Works great.
Thanks !
We were all desesperated !

Thierry-Myron, 2021-07-08:
Works like a charm !
Many thanks.