BillyBlaze/OctoPrint-TouchUI #496: local browser (installed using touchui-autostart) does not connect to touchui

Nailig, 2021-05-14:
I ran the touchui-autostart script
touchui starts, browser starts and tries to connect
browser can not connect.

My solution:
in /etc/default/touchui 
replace 
TOUCHUI_PORT="8888"
with 
TOUCHUI_PORT=""

octopi 1.5.x and 1.6.x
touchui 0.3.17

raster, 2021-05-19:
@Nailig Thanks! This fixed my connection issue.

kazolar, 2021-05-26:
this has to do with the script.js -- it's in the TouchUI-autostart/bootloader folder -- change the port to "80" and url to "http://localhost" -- then the bootloader works fine

orrinwitt, 2021-06-21:
I had the exact same issue as @Nailig and deleting 8080 fixed it for me also.

zpatten, 2021-06-22:
Hit this issue as well; I'm running OctoPrint bound to localhost only with haproxy in front of it; had to set `TOUCHUI_PORT=""` in order to get TouchUI to load on the Pi's touchscreen.

kazolar, 2021-06-24:
Setting the config port to 8080 bypasses the bootloader -- so if it takes a few seconds for octoprint to start, touchui will not start properly, the fix should be in bootloader -- i changed the url to "http://localhost" in script.js in the bootloader directory. That fixes the bootloader and touch ui boot script works as intended.

zpatten, 2021-06-24:
Not sure what you're getting at.  Changing script.js broke everything for me, it just says it can't connect and dies; I've had zero issues since changing the port.  The OP references the auto start bits.  The port is `8888` not `8080`.

- https://github.com/BillyBlaze/OctoPrint-TouchUI-autostart/blob/master/touchui.default#L13

kazolar, 2021-06-24:
It was not loading for me -- yes I meant port 8888 -- I left it at 8888 and I changed the 
script.js 
I made the following changes:
        var port = "80";
        var url =  "http://localhost";

By removing 8888 you are bypassing bootloader and connecting directly to octoprint -- which does work, but is not the intention -- the intention is to have the TouchUI loading screen on startup it pings octoprint and forwards to it after it loads.

zpatten, 2021-06-25:
Thanks for the details; I'll give those changes a try then :+1:

zpatten, 2021-06-29:
After some testing and trying your suggestion it doesn't seem to work at all.  Changing script.js breaks everything for me.  It always results in a connection reset error in the browser.

Nailig, 2021-06-29:
@zpatten just use my first solution, it's simple and seems to work for many people. You should not edit the code file if the fix is just to use the correct config. port 8080 used to work in the past, but that changed. Remove the portnumber and it will default to the correct port.

kazolar, 2021-06-29:
bypassing 8888 is not using the bootloader, bootloader is obviously broken, if you can connect a keyboard to the pi, you can go pop up F12 (should work) and you can tell where the bootloader gets stuck, for me removing js calculations for host and port fixed it, but may not work for everyone. If not using bootloader works for you -- then fine, still means the bootloader is broken

Nailig, 2021-06-29:
@kazolar your fix breaks things for people, my fix make things work for people. Do you have more changes to make it work for you?

And.... If my fix bypasses the bootloader and everything works just fine, then what is the function of the bootloader? Then we can just delete the bootloader. I think something else is the problem,

kazolar, 2021-06-29:
My fix to the bootloader restored its behavior on my 3 octopi machines. Turning it off by removing 8888 bypasses the Javascript bug, and then touch ui just starts up without being started by the bootloader. I had fixed it on my printers months ago. There are other ways to fix the bootloader by hard coding the pi port/ip. Bootloader purpose may have outlived its usefulness, it is supposed serve as a splashscreen while octoprint starts up, but if you're using a fast enough sbc you won't miss it.

scyto, 2021-06-30:
brand new octoprint user, brand new touchui user

changing the config file seems like the right approach to me (not editing js script files that can change on upgrades).
On my config it was ""8888 and had to be changed to ""

I see no evidence on my machine that anything is listening on port 8080 or 8888 - it seems the octorpint OS image run everything on 80 and 443 behind  ha proxy. 

@Nailig thanks for figuring this out; this seems like the right workaround - though seems like this needs to be fixed in some way, not sure who the prohject maintainer is?

zpatten, 2021-07-01:
Ya this was my conclusion as well.  Regardless of changing script.js or not; the server that is supposed to provide it doesn't start up.  This is why if you leave the port at `8888` the browser just dies.  In my case it would show a connection reset, which is what would happen if that process tried to start up and then died off.  I see where it is supposed to be executed but it seems to be dead/broken code as the python server it's supposed to start is never running.  I don't have the time to figure out why it is broken and changing the port so it just hits octoprint works great and I've never had it fail on any of my 4 installations.

BillyBlaze, 2021-07-04:
Thanks for reporting. The issue has been fixed.

If you made changes to your default file then restore `8888`
If you changed the script.js file then restore the original files by running `git reset --hard` in the TouchUI-autostart folder.

After that you can [update your bootloader](https://github.com/BillyBlaze/OctoPrint-TouchUI/wiki/Setup:-Update-your-bootloader) as described in the wiki.

closed 2021-07-04 by BillyBlaze

zpatten, 2021-07-05:
@BillyBlaze Thanks!  Appreciate your project and fixing this!

zpatten, 2021-07-05:
Happy to report so far everything is working perfect and the python server is functioning properly now along with the bootloader.