BillyBlaze/OctoPrint-TouchUI #497: When "Enable Customization" is enabled tab bar no longer appears

beelsebob, 2021-05-15:
When "Enable Customization" is enabled the tab bar at the top of the screen no longer presents any buttons.  Instead it becomes a solid coloured header bar, and controls disappear under it.

This occurs even with a fresh install of TouchUI with no customisations specified.

theHalo, 2021-05-19:
Same Problem here. Just a blank, blue bar at the top. Switching off customizations reverts to desktop mode.

avigeilpro, 2021-05-20:
> 
> 
> Same Problem here. Just a blank, blue bar at the top. Switching off customizations reverts to desktop mode without any tab icons either.

after switching off customization you have to restart octoprint, and all will be back.

theHalo, 2021-05-21:
That did the trick. Thanks for the hint.

Jahislove, 2021-05-23:
same problem here . only blue bar at top without tab icons
switching customization off and restarting octoprint did the job

devildant, 2021-06-02:
@BillyBlaze still no solution?

devildant, 2021-06-02:
link to https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/494

beelsebob, 2021-06-17:
Bumping, because this issue is still valid.

closed 2021-06-25 by github-actions[bot]

BillyBlaze, 2021-07-04:
The issue has been fixed on the maintenance branch and will be released a.s.a.p

closing this issue since its a duplicate of #494