BillyBlaze/OctoPrint-TouchUI #498: auto login not working

jgillis09, 2021-05-15:
It seems as if when i ran installer on pi- it did not add "autologin" entries into the .yaml file .

every time i try and manually add them and local network entries octoprint will fail to load- not accessible from pc browser or from touchui on pi

I've tried to copy past the log request in other threads but i get an error that i dont have permission to view the logs? dunno who else would. 

Help me if you can- i'll get whatever information i can for you . Thanks

jgillis09, 2021-05-15:
update- i have the autologin commands in my .yaml file and they have not broken octoprint as a whole- now however; touchui still fails to load- it shows the loading screen for a while- then a white screen with yeild sign and error: browsers error console with a link to octoprint.log. then "startup failed" screen

current logs: 
[octoprint.log](https://github.com/BillyBlaze/OctoPrint-TouchUI/files/6488023/octoprint.log)

kazolar, 2021-05-26:
see the fix i did in https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/496 -- i had the same issue.

closed 2021-06-18 by github-actions[bot]