BillyBlaze/OctoPrint-TouchUI #500: Can't select menu items below the viewport

leanazulyoro, 2021-06-25:
Hi! I noticed difficulties selecting elements in the menu that I need to scroll in order to reach.

Here, a video is worth a million words:
https://user-images.githubusercontent.com/1864435/123491568-5a119800-d5ed-11eb-8c26-9c7b3c363c88.mov

This happens to me on any smartphone I have (all android devices), as well as desktop emulating a mobile device with touch events. It does not happen however without touch emulation:
https://user-images.githubusercontent.com/1864435/123491703-bd032f00-d5ed-11eb-855b-eed3ffc8067a.mov

BillyBlaze, 2021-07-04:
Hi! Thanks for reporting. I have looked at the video's and I have tried to reproduce the same. However I am unable to get the same issue with both the Firefox emulation and my android device.

My suspicion is that a plugin is causing a conflict. Could you perhaps share your installed plugins? or you could disable them to see which one is the culprit and report the issue over there.

leanazulyoro, 2021-07-04:
I tried disabling all my plugins, and it worked ok, but I suspect it's only because having them disabled makes the menu shorter, and that way it actually fits in the viewport. I then installed plug-ins a few at a time until they populated the menu enought so it needs scrolling to reach the last items. At that moment the elements that are outside the first scroll are not tappable.

Here I need to scroll in order to reach the logout and system items, and indeed, they present the bug.
[image]

leanazulyoro, 2021-07-04:
I can even force the bug to occur on landscape mode, and that way it happens on higher items, since there are fewer that fit in the menu.

I can tap all those items visible
[image]

But I cannot tap the ones under the first scroll
[image]

leanazulyoro, 2021-07-04:
Ok, I'm pretty confident the spool manager plugin is the culprit. Please let me know if you can reproduce it by enabling that plugin!
[image]

gobbledygoober, 2021-07-05:
I encountered this as well. 

OctoPrint 1.6.1
Python 3.7.3
OctoPi 0.18.0
TouchUI 0.3.17
SpoolManager 1.4.3

@leanazulyoro - nice find! Your workaround of disabling SpoolManager worked for me as well. 

It should be noted that I only installed SpoolManager because it is listed as an optional/recommended plug-in when installing PrintJobHistory (1.14.0).

BillyBlaze, 2021-07-05:
Thanks for the information. I am able to reproduce the issue too now.

The good news: 
I have tracked down the code that is interfering. Inside SpoolManager there is a plugin named 'pick-a-color.js'. [This piece of code](https://github.com/OllisGit/OctoPrint-SpoolManager/blob/master/octoprint_SpoolManager/static/js/pick-a-color.js#L775-L793) called `executeUnlessScrolled` (its even in the name) is preventing the items to be clicked after scrolling on a touch device. All I can say is 'bad plugin'!

The bad news:
I cannot do anything about it. It needs to be fixed inside the code base of SpoolManager. The authors of SpoolManager might even need to report this to 'pick-a-color.js'.

----

As there is nothing I can do inside TouchUI I am therefor closing this issue. Please feel free to create an issue in SpoolManager and refer to this issue for some more technical details.

closed 2021-07-05 by BillyBlaze

leanazulyoro, 2021-07-05:
@BillyBlaze thanks for the help! I will report to spoolmanager developers, as I did want to use it to track my spools.

gobbledygoober, 2021-07-05:
@leanazulyoro are you planning on posting on https://github.com/OllisGit/OctoPrint-SpoolManager/issues?

gobbledygoober, 2021-07-05:
> @leanazulyoro are you planning on posting on https://github.com/OllisGit/OctoPrint-SpoolManager/issues?

Since you logged this issue, I logged the SpoolManager one here:

https://github.com/OllisGit/OctoPrint-SpoolManager/issues/188

Thanks!

leanazulyoro, 2021-07-05:
> > @leanazulyoro are you planning on posting on https://github.com/OllisGit/OctoPrint-SpoolManager/issues?
> 
> Since you logged this issue, I logged the SpoolManager one here:
> 
> https://github.com/OllisGit/OctoPrint-SpoolManager/issues/188
> 
> Thanks!

That's perfect, yes I was gonna post it there, feel free to tag me there if you need anything that would help solve the issue.